prusa3d/PrusaSlicer #8543: Arachne uses inconsistent number of perimeters on consistent width model

dnew92130, 2022-07-21:
### Description of the bug

Using 2.5.0-alpha 2 or 3, turning on arachne, slicing tall cube with no top.
The number of perimeter lines change, from one thick to two thin, around the 55mm height. This doesn't seem right to me.

I sliced walls with a constant thickness with Arachne turned on.

I expected Arachne to generate a consistent pattern of perimeter moves.

Instead, around 55mm high, the pattern of perimeters changes.

Includes project file, original STL, photos of printed object that alerted me to the problem. (And boy was *that* fun figuring out what was wrong. ;-)

### Project file & How to reproduce

[DNewIssue.zip](https://github.com/prusa3d/PrusaSlicer/files/9162595/DNewIssue.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-alpha3+win64-202207151230

### Operating system

Windows 10 Pro x64 (up to date)

### Printer model

Prusa MINI+

hejllukas, 2022-07-22:
I looked at your issue, and I don't think that this is the issue of Arachne. The number of perimeters is changing because the hole inside the model is wider on the bottom than on the top. I'm attaching screenshots from Blender with absolute vertex position.

Bottom                    |  Top
:-------------------------:|:-------------------------:
[image]  |  [image]

The wall thickness on the bottom is 0.57735mm, but on the top is 0.70708mm.
So the width of the model walls slightly increases with layer height. And there is a place when the model is so thick that Arachne places two perimeters instead one extremely thick perimeter.
So for this model and configuration, the behavior of Arachne is correct. Also, when you use the Classic perimeter generator same behavior is there, but at a different layer height.

But suppose you want to print models like this with a single perimeter. In that case, you have to, for example, change the extrusion width for `External perimeters` to something like 0.75mm (should be still printable with a 0.4mm nozzle) or similar.

dnew92130, 2022-07-22:
I did not realize Blender would make the walls different thicknesses with the Solidify modifier. I told Blender to use an even thickness and the problem disappeared. My apologies for wasting your time. :-)

closed 2022-07-22 by dnew92130