prusa3d/PrusaSlicer #8547: Monitor STL file for changes

dcorbin, 2022-07-23:
**Is your feature request related to a problem? Please describe.**
No, not a software problem, but a process problem.

**Describe the solution you'd like**
Have an option that would automatically re-import an STL file when the STL file is modified.

**Describe how it would work**
Have a checkbox (presumably in Print Settings) that I can check.   Let's call it Auto ReImport
Remember with each object the file that it came from.
Use the operating system features to be notified if that file changes.  When notified, replace the imported object(s) on the Plater with the updated ones

**Additional context**
A _lot_ of 3D design is very interative. 
1) Design STL
2) Import
3) Slice
4) Print
5) Repeat.

Repeat usually (for me) is delete the object on the virtual Plate, and re-import.   Anything that can be done to eliminate manual steps from this process can be huge boon.  Note: sometimes the object has to go through some transformations before printing (position and orientation) -- remembering that auto-applying would be even better.

neophyl, 2022-07-23:
Why aren’t you using the reload option ?  Right click on an updated object and the option should be available which does remember some if not all transformations, but doesn’t do things like variable layer heights applied as those get reset.

dcorbin, 2022-07-23:
Ignorance of it? I'll do that, but I stil think auto-reloading would be a nice feature.

mtaumike, 2022-07-26:
If this were implemented I'd suggest making it a preference, default to `False`

Edge case: Shared filesystem, multiple folks working on files. You have STL opened and working on slicing and tuning, teammate working on the STL, an autosave occurs, and there was a problem. Now you have a corrupted or bad STL.

Making it a preference will help those in multi-designer/user set ups.

KimmoHop, 2022-07-28:
Not sure if PS had  in the past, or am I remembering Cura, notification about changed file with option to reload it?

0x6d64, 2023-01-03:
> Not sure if PS had in the past, or am I remembering Cura, notification about changed file with option to reload it?

This would be pretty nice: a small icon (e.g. a circle arrow) appearing the object list whenever a changed file is detected, a click on that circle triggers the "reload STL" functionality.

dcorbin, 2026-09-02:
Still worthwhile in some form, IMO.