prusa3d/PrusaSlicer #8548: Arrange Rotation not working with parts larger that build volume along one axis

SkeletalPrints, 2022-07-23:
### Description of the bug

Arrange Rotation does not work with parts larger than the build volume along one axis, and that still fits inside the build volume when rotated at an angle.

Expected:
[image]
What is dose:
[image]

### Project file & How to reproduce

[Arange Rotation Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/9174343/Arange.Rotation.Bug.zip)

Open file
Click arrange tool

I tried it with a box and it does not work if the box is 260 on x
[image]
But if it is 260 on y it does work
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3s

dartrax, 2022-07-24:
Also discussed in #5339

stepikovo, 2022-07-24:
I'm closing this as a mentioned duplicate.

closed 2022-07-24 by stepikovo