prusa3d/PrusaSlicer #8552: 2.4.0+x64 on Mac 10.13.6 causes lag throughout system

tmorris9, 2022-07-24:
### Description of the bug

Lag on all software when PrusaSlicer is open.

### Project file & How to reproduce

Simply open PrusaSlicer, slice a model then leave window open. When you go to any other software while slicer is in background there is noticeable lag through the system. Mouse movements, scrolling, screen refresh.... 

Once the slicer is closed the machine instantly becomes responsive again with no lingering effects.

When I work on projects I often want to jump back and froth from my modeling software to slicer and this can cause issues.

File does not matter, will do this with any file no matter the complexity or file size.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

Mac OS 13.6.0

### Printer model

MK3S

stepikovo, 2022-07-25:
Can you please try the newest version?

Can you please provide the system info? (Help - System info)

tmorris9, 2022-07-26:
Hello,

OK, I got the new version and will do some testing to see if it’s still an issue.

PrusaSlicer
Version:   2.4.0+x64
Build:     PrusaSlicer-2.4.0+x64-202112211607

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS High Sierra Version 10.13.6 (Build 17G14042)
Total RAM size [MB]: 42,950MB
OpenGL installation
GL version:   2.1 ATI-1.68.25
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 580 OpenGL Engine
GLSL version: 1.20
<details>
<summary>Installed extensions:</summary>
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix
</details>

> 
> 
> Can you please try the newest version?
> 
> Can you please provide the system info? (Help - System info)

Terry Morris
2645 Wilson St. 
Carlsbad, CA 92008

tmorris9, 2022-07-26:
OK, using the newest version for a few hours I have not had the issue. I cannot say for 100% sure it’s fixed yet but I am fairly sure it is.

Thanks

> 
> 
> Can you please try the newest version?
> 
> Can you please provide the system info? (Help - System info)

Terry Morris
2645 Wilson St. 
Carlsbad, CA 92008

tmorris9, 2022-07-28:
OK, after a full 2 days of using the computer with the Alpha build I can say I have not had the issue (even forgot it was still running in the background).

> 
> 
> Can you please try the newest version?
> 
> Can you please provide the system info? (Help - System info)

Terry Morris
2645 Wilson St. 
Carlsbad, CA 92008

tmorris9, 2022-10-11:
Well, it turns out the problem still exists, but mainly with larger models now.

> 
> 
> Can you please try the newest version?
> 
> Can you please provide the system info? (Help - System info)

Terry Morris
2645 Wilson St. 
Carlsbad, CA 92008

oskarsss, 2023-06-05:
I have similar issue, that whenever I open slicer, my filesystem just stops to work properly (even via commandline). I am using linux arch core 5.13.0-valve36-1-neptune (steamdeck) and installed it via flatpak. I kinda solved this problem when took permisions from app (using flatseal) of all user files and /run folders access. Now it works ok (hopefully it will do that further) but previously it sliced model in seconds and the I waited for 2-4 minutes to export. Also were unable to open other programs (like browser or cura for tests). Rven closing app did not help, only reboot did

oskarsss, 2023-06-05:
Update: narroved down (threw few reboots) to /run/media access causes lags. Whenever I disable it everything works fine.

lukasmatena, 2023-06-05:
Guys, we are grateful for the reports, but it is essential that you share at least the version of PrusaSlicer that you are using. Ideally try in 2.6.0-beta4.

@oskarsss Does the appimage downloaded from our github suffer from the same problem?

closed 2026-09-29 by prusaslicer-bot[bot]