prusa3d/PrusaSlicer #8553: Can the ironing junction be extended to the inside

xiaogui8dangjia, 2022-07-24:
[image]
[image]
Ironing will extrude excess material, which will accumulate in the corners of the model and affect the print quality

xiaogui8dangjia, 2022-07-24:
When ironing, it can be extended to the inside of the model, which can eliminate the excess material outside the model

xiaogui8dangjia, 2022-07-24:
[image]

stepikovo, 2022-08-04:
Can you please try printing it using the latest 2.5.0-alpha3 version with the **[pressure equalizer](https://help.prusa3d.com/article/pressure-equalizer-_331504)**? I believe it could help and solve the issue.

closed 2026-09-29 by prusaslicer-bot[bot]