prusa3d/PrusaSlicer #8559: Feature Request: Add "Nozzle Name" field in printer settings

FIRES-3D, 2022-07-25:
**Feature Description:**
Add a "Nozzle Name" or "Tool Name" field near "Nozzle Diameter" in the Printer Settings tab. 

**Describe the solution you'd like:**
A "Nozzle Name" field would allow the user to (if desired) provide a unique identifier for the nozzle used in that profile. This may include details such as nozzle material, brand, etc. This field should be accessible in G-Code sections via a [nozzle-name] handle or similar. 

Functionality gained:
- Make the nozzle material / type easily identifiable when sharing profiles between users. Currently the only nozzle "identifier" is the diameter
- Reduce confusion for those who have many different types of nozzles of the same size (e.g. brass vs steel vs tungsten carbide etc)
- Post-processors and firmware macros can be written to take different actions based on [nozzle-name]. For example, a user can automatically load custom nozzle offsets for each nozzle at the start of a print. 
- Allow Z-offset to be "offloaded" to firmware (as in the example above). The slicer Z-offset can be left at 0, making it far easier to share the 3MF profile (or even the G-Code) between users/machines without accidental crashes. 
- If Z-offsets ever change, the user does not have to go back and re-update all of their profiles and re-slice all their files, they can simply update the post-processor / firmware calibration values (all while using the same unchanged G-Code / profiles)

This functionality can be patched together via manual G-Codes, but:
- This obfuscates the identifying information, as it would be buried in blocks of G-Code
- Unique IDs stored in G-Code would need to be manually updated, making each profile's G-Code unique (as opposed to just [nozzle-name] handle). It's preferable that all G-Code sections be identical so any updates are easy to proliferate.

stepikovo, 2022-07-26:
Can't this be done by creating a custom printer profile with the type of the nozzle in the name of the profile?

FIRES-3D, 2022-07-27:
> Can't this be done by creating a custom printer profile with the type of the nozzle in the name of the profile?

Partially - this is actually the solution I'm currently working with. It's workable, but a bit clunky and not easily extensible for firmware macros / post-processors (Klipper in my case). A [nozzle_name] field could more easily be referenced in the "Start G-Code" while still accommodating any naming convention the user desires (simpler and shorter profile names). 

Identifying nozzle type by parsing the profile name also creates more "accounting" work, as macros / post-processors need updating each time profile names or naming conventions change.

closed 2026-09-29 by prusaslicer-bot[bot]