prusa3d/PrusaSlicer #8561: Change in extrusion width on single perimeter part on Arachne

sundewzer, 2022-07-25:
### Description of the bug

I have a part that is only two perimeters thick, or another way to say it is only an external perimeter. This is an odd case because I have the extrusion width quite high. In classic I get a width of 1.31 and in Arachne I get 1.17. When it prints the parts in Arachne there is no overlap and the walls separate. I realize is is a special case and won't effect most people but I thought it was worth mentioning. 

### Project file & How to reproduce

That part is 2.29mm wide and have the external perimeter width set to 1.31

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Win 10

### Printer model

Voron Trident and MK3

stepikovo, 2022-07-26:
Can you please upload your project file?

sundewzer, 2022-07-26:
[voron cht reducer v2.zip](https://github.com/prusa3d/PrusaSlicer/files/9191241/voron.cht.reducer.v2.zip)
Sure thing.

hejllukas, 2022-09-01:
This issue is similar to #8578, so I cut important information from the answer I posted there.

It should be mentioned that the Classic perimeter generator and the Arachne perimeter generator work differently with entered extrusion width. With Classic perimeter generator, whatever extrusion width you give it, it uses it and doesn't check if the object is over-extruded, that's up to the user to ensure. The slicer will only recommend to the user which wall thickness the object should have for entered extrusion width (shown below the perimeters input field).
In general, the Classic perimeter generator over-extrudes in a lot of cases. Because entered extrusion width will be used everywhere, even if it leads to over-extrusion.
Over-extrusion occurs especially in objects with only two internal perimeters or where perimeters from different object walls meet (for example, a hole in an object near another wall)
This means that the Classic perimeter generator, in many cases, can hide under-extrusion issues (caused by bad extruder calibration, wrong filament diameter, wrong extrusion multiplier, too fast printing, hotend unable to melt the specified amount of material, extrusion too wide for used nozzle diameter, extruder gears slipping, and much more).

On the other hand, Arachne doesn't over-extrude because it just takes the specified extrusion width as some initial value and calculates extrusion width according to the object wall thickness at every place. So that the object is filled with the exact amount of material, with no over-extrusion or under-extrusion caused by a slicer.
But this means that if a printer has any under-extrusion or over-extrusion issues, Arachne amplifies those issues compared to Classic, which will be your case.

You are usign quite extereme configuration. I think that printing a 1.31mm extrusion line with a 0.6mm nozzle is quite extreme, and I believe that the real extruded perimeter will not have 1.31mm but much less, mainly because of the backpressure that extruding of 1.31mm through 0.6mm nozzle is causing.
Otherwise, with such a large extrusion width, a 2.29mm thick object will be quite over-extruded because there will be a 0.33mm overlap if a printer really prints an extrusion line with this width.

I checked the G-Code. And extrusion width is mostly around 1.175. So based on what I wrote above, Arachne adjusts extrusion width based on an object's wall thickness and expects that the printer can extrude it without under-extrusion. So based on this, Arachne uses an extrusion width of around 1.175, so there should be around 0.06mm overlap (that should be enough if a printer can extrude this extrusion value).

You can also try to print an object with just one perimeter with the same speed (generally with the same settings) as in the attached 3MF with an extrusion width of 1.31mm and the Classic perimeter generator. And then, measure perimeter width using a micrometer or caliper (with sufficient measure accuracy), and you will see that width of this perimeter will not be 1.31mm, but smaller.

So the issue will be caused that Arachne is expecting that the printer is able to extrude line 1.175 width, but in reality, the printed line will be thinner, mainly because you're trying to print a very wide line through a relatively small nozzle (compared to extrusion width).

closed 2022-09-01 by hejllukas