prusa3d/PrusaSlicer #8563: Segmentation Fault on Debian

jcress, 2022-07-25:
### Description of the bug

Running prusa slicer fails with 'Segmentation Fault'

I have tried the 2.4 appimages and 2.5 alpha app image (GTK2 and GTK3)

I tried building from source. 
```

jcress@batcave:~/Downloads$./PrusaSlicer-2.5.0-alpha3+linux-x64-GTK3-202207151254.AppImage --loglevel 5
[2022-07-25 15:41:25.464521] [0x00007fd384b7a8c0] [trace]   Initializing StaticPrintConfigs
[2022-07-25 15:41:25.540145] [0x00007fd384b7a8c0] [debug]   full path: 8124743336197116828
[2022-07-25 15:41:25.540161] [0x00007fd384b7a8c0] [debug]   single instance: undefined. other params: /tmp/.mount_PrusaSnuyp3G/usr/bin/bin/prusa-slicer;--loglevel;5
[2022-07-25 15:41:25.540165] [0x00007fd384b7a8c0] [debug]   full lock path: /home/jcress/.config/PrusaSlicer-alpha/cache/8124743336197116828.lock
[2022-07-25 15:41:25.540215] [0x00007fd384b7a8c0] [debug]   Creating lockfile.
[2022-07-25 15:41:25.540221] [0x00007fd384b7a8c0] [info]    Instance check: Another instance not found or single-instance not set.
[2022-07-25 15:41:25.594691] [0x00007fd384b7a8c0] [trace]   System language detected (user locales and such): en_US
[2022-07-25 15:41:25.595907] [0x00007fd384b7a8c0] [trace]   Best translation language detected (may be different from user locales): en_US
[2022-07-25 15:41:25.595922] [0x00007fd384b7a8c0] [trace]   Switching wxLocales to en_US
[2022-07-25 15:41:28.906739] [0x00007fd384b7a8c0] [info]    Checking if indices need to be installed from resources...
[2022-07-25 15:41:28.917111] [0x00007fd384b7a8c0] [info]    Update index from resources: "PrusaResearch.idx"
[2022-07-25 15:41:28.917128] [0x00007fd384b7a8c0] [debug]   PresetUpdater: Copying "/tmp/.mount_PrusaSnuyp3G/usr/bin/bin/../resources/profiles/PrusaResearch.idx" -> "/home/jcress/.config/PrusaSlicer-alpha/cache/PrusaResearch.idx"
[2022-07-25 15:41:28.917149] [0x00007fd384b7a8c0] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): No such file or directory
[2022-07-25 15:41:29.232425] [0x00007fd384b7a8c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 82MB; Shared memory: 53MB; Private memory: 29MB; Peak memory usage: 82MB
[2022-07-25 15:41:29.233991] [0x00007fd364f3e700] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object8124743336197116828 registered. Starting listening for messages.
[2022-07-25 15:41:29.344983] [0x00007fd384b7a8c0] [debug]   BuildVolume bed_shape clasified as: Rectangle
Segmentation fault

```

### Project file & How to reproduce

I can't create a project file. This happens before prusa slicer loads. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha3+linux-x64-GTK3

### Operating system

Debian 11

### Printer model

Prusa Mini

jcress, 2022-07-25:
```
`jcress@batcave:~/code/PrusaSlicer/build/src (master)$./prusa-slicer --loglevel 5
[2022-07-25 15:47:16.614830] [0x00007efe55ed0880] [trace]   Initializing StaticPrintConfigs
[2022-07-25 15:47:16.632331] [0x00007efe55ed0880] [trace]   App updater default download path: "/home/jcress/Downloads"
[2022-07-25 15:47:16.632413] [0x00007efe55ed0880] [debug]   full path: 1269281012993219591
[2022-07-25 15:47:16.632429] [0x00007efe55ed0880] [debug]   single instance: undefined. other params: ./prusa-slicer;--loglevel;5
[2022-07-25 15:47:16.632441] [0x00007efe55ed0880] [debug]   full lock path: /home/jcress/.config/PrusaSlicer-alpha/cache/1269281012993219591.lock
[2022-07-25 15:47:16.632473] [0x00007efe55ed0880] [debug]   Creating lockfile.
[2022-07-25 15:47:16.632485] [0x00007efe55ed0880] [info]    Instance check: Another instance not found or single-instance not set.
[2022-07-25 15:47:16.668841] [0x00007efe55ed0880] [trace]   System language detected (user locales and such): en_US
[2022-07-25 15:47:16.669343] [0x00007efe55ed0880] [trace]   Best translation language detected (may be different from user locales): en_US
[2022-07-25 15:47:16.669354] [0x00007efe55ed0880] [trace]   Switching wxLocales to en_US
[2022-07-25 15:47:20.254491] [0x00007efe55ed0880] [info]    Checking if indices need to be installed from resources...
[2022-07-25 15:47:20.410207] [0x00007efe55ed0880] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 88MB; Shared memory: 58MB; Private memory: 29MB; Peak memory usage: 88MB
[2022-07-25 15:47:20.411690] [0x00007efe30f3e700] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object1269281012993219591 registered. Starting listening for messages.
[2022-07-25 15:47:20.502190] [0x00007efe55ed0880] [debug]   BuildVolume bed_shape clasified as: Rectangle
Segmentation fault
jcress@batcave:~/code/PrusaSlicer/build/src (master)$
`
```

lukasmatena, 2022-07-26:
@jcress Thanks. Are you able to provide a backtrace using gdb?

AntoineJeanson, 2022-07-26:
Also experiencing this on Manjaro, although logging stops earlier. Should I create another issue ? Thanks for your help.

```
$ prusa-slicer --loglevel 5
[2022-07-26 14:31:44.290507] [0x00007fd6133ead80] [debug]   full path: 14269235640737936292
[2022-07-26 14:31:44.290521] [0x00007fd6133ead80] [debug]   single instance: undefined. other params: prusa-slicer;--loglevel;5
[2022-07-26 14:31:44.290524] [0x00007fd6133ead80] [debug]   full lock path: /home/antoinej/.config/PrusaSlicer/cache/14269235640737936292.lock
[2022-07-26 14:31:44.290532] [0x00007fd6133ead80] [debug]   Creating lockfile.
[2022-07-26 14:31:44.290534] [0x00007fd6133ead80] [info]    Instance check: Another instance not found or single-instance not set.
[2022-07-26 14:31:44.320403] [0x00007fd6133ead80] [trace]   System language detected (user locales and such): en_US
zsh: segmentation fault (core dumped)  prusa-slicer --loglevel 5
```

bubnikv, 2022-07-26:
We need you to run PrusaSlicer under debugger and provide us what the debugger tells you when PrusaSlicer crashes.
run
$ gdb prusa-slicer
then print
run
in gdb

jcress, 2022-07-26:
`Starting program: /home/jcress/code/PrusaSlicer/build/src/prusa-slicer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2022-07-26 16:27:02.909840] [0x00007ffff5786880] [trace]   Initializing StaticPrintConfigs
[Detaching after fork from child process 161033]
[New Thread 0x7ffff5423700 (LWP 161034)]
[New Thread 0x7ffff4c22700 (LWP 161035)]
[Detaching after vfork from child process 161036]
[Detaching after vfork from child process 161038]
[Detaching after vfork from child process 161040]
[Detaching after vfork from child process 161042]
[Detaching after vfork from child process 161044]
[Detaching after vfork from child process 161046]
[Detaching after vfork from child process 161048]
[New Thread 0x7fffeffff700 (LWP 161050)]
[New Thread 0x7fffef7fe700 (LWP 161051)]
[Thread 0x7fffef7fe700 (LWP 161051) exited]
[New Thread 0x7fffef7fe700 (LWP 161053)]
[New Thread 0x7fffeee1f700 (LWP 161055)]
[New Thread 0x7fffee61e700 (LWP 161056)]
[New Thread 0x7fffede1d700 (LWP 161057)]
[New Thread 0x7fffecc92700 (LWP 161058)]
[New Thread 0x7fffd9f40700 (LWP 161059)]
[New Thread 0x7fffd973f700 (LWP 161060)]

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x0000555556675cc8 in Slic3r::GUI::GLTexture::load_from_svg_files_as_sprites_array(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::pair<int, bool>, std::allocator<std::pair<int, bool> > > const&, unsigned int, bool) ()
(gdb) Quit
`

jcress, 2022-07-27:
based on what I read [here](https://github.com/supermerill/SuperSlicer/issues/854), I tried switching my window manager. 

The prusa-slicer I built from source works fine on Gnome.  The segmentation fault I am reporting above happens when I use i3 (my default WM)

chrishoage, 2022-08-21:
I am having the same issue on Arch with BSPWM. 

However - the app image worked for a while. I was able to launch and use it 3 times - after that it stopped working. 

I tried moving the `.config/PrusaSlicer` away and starting fresh but the problem presisted. 

```
(gdb) run
Starting program: /home/chris/Downloads/PrusaSlicer-2.5.0-rc1+linux-x64-GTK3-202208191449.AppImage
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 107096]
process 107093 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 107101]
[Detaching after fork from child process 107103]
process 107093 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 107104]
process 107093 is executing new program: /tmp/.mount_PrusaSPMl4Ct/usr/bin/bin/prusa-slicer
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[2022-08-21 12:04:31.082310] [0x00007ffff317f7c0] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7ffff2d3c6c0 (LWP 107118)]
[New Thread 0x7ffff253b6c0 (LWP 107119)]
[Detaching after vfork from child process 107120]
[Detaching after vfork from child process 107122]
[Detaching after vfork from child process 107124]
[Detaching after vfork from child process 107126]
[New Thread 0x7ffff1ba86c0 (LWP 107128)]
[Thread 0x7ffff1ba86c0 (LWP 107128) exited]
[New Thread 0x7ffff1ba86c0 (LWP 107129)]
[New Thread 0x7ffff0bd76c0 (LWP 107130)]
[New Thread 0x7fffe3fff6c0 (LWP 107131)]
[Thread 0x7ffff0bd76c0 (LWP 107130) exited]
[Thread 0x7fffe3fff6c0 (LWP 107131) exited]
[New Thread 0x7fffe3fff6c0 (LWP 107132)]
[New Thread 0x7ffff0bd76c0 (LWP 107133)]
[Thread 0x7fffe3fff6c0 (LWP 107132) exited]
[New Thread 0x7fffe3fff6c0 (LWP 107134)]
[Thread 0x7ffff0bd76c0 (LWP 107133) exited]
[New Thread 0x7ffff0bd76c0 (LWP 107135)]
[Thread 0x7fffe3fff6c0 (LWP 107134) exited]
[New Thread 0x7fffe3fff6c0 (LWP 107136)]
[Thread 0x7ffff0bd76c0 (LWP 107135) exited]
[New Thread 0x7ffff0bd76c0 (LWP 107137)]
[Thread 0x7fffe3fff6c0 (LWP 107136) exited]
[Thread 0x7ffff0bd76c0 (LWP 107137) exited]
[New Thread 0x7ffff0bd76c0 (LWP 107138)]
[New Thread 0x7fffe3fff6c0 (LWP 107139)]
[Thread 0x7ffff0bd76c0 (LWP 107138) exited]
[Thread 0x7fffe3fff6c0 (LWP 107139) exited]
[New Thread 0x7fffe3fff6c0 (LWP 107140)]
[New Thread 0x7ffff0bd76c0 (LWP 107141)]
[Thread 0x7fffe3fff6c0 (LWP 107140) exited]
[Thread 0x7ffff0bd76c0 (LWP 107141) exited]
[New Thread 0x7ffff0bd76c0 (LWP 107143)]
[New Thread 0x7fffe3fff6c0 (LWP 107144)]
[New Thread 0x7fffe37fe6c0 (LWP 107145)]
[New Thread 0x7fffe2ffd6c0 (LWP 107146)]
[New Thread 0x7fffe11ff6c0 (LWP 107147)]
[New Thread 0x7fffe09fe6c0 (LWP 107148)]
[New Thread 0x7fffcd9ff6c0 (LWP 107149)]
[New Thread 0x7fffcd1fe6c0 (LWP 107152)]
[Thread 0x7ffff0bd76c0 (LWP 107143) exited]
[Thread 0x7fffe37fe6c0 (LWP 107145) exited]
[Thread 0x7ffff1ba86c0 (LWP 107129) exited]
[Thread 0x7fffe2ffd6c0 (LWP 107146) exited]
[New Thread 0x7fffe2ffd6c0 (LWP 107173)]

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00000000010f8f88 in Slic3r::GUI::GLTexture::load_from_svg_files_as_sprites_array(std::vector<std::string, std::allocator<std::string> > const&, std::vector<std::pair<int, bool>, std::allocator<std::pair<int, bool> > > const&, unsigned int, bool) ()
(gdb) quit
```

chrishoage, 2022-08-21:
I discovered that this seg fault is application window size dependent. 

If I launch Prusa Slicer such that it is not tiled - or has a great deal of space - there is no seg fault.

bubnikv, 2022-08-22:
@jcress Thanks for the call stack, we will investigate.

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Let us know if problem persists. Closing.

closed 2023-02-01 by lukasmatena