prusa3d/PrusaSlicer #8566: Lead In Tail

adamfilip, 2022-07-26:
When starting a print, sometimes a blob of filament will form on the tip of the nozzle as its starts to ooze out, Often we will print a skirt or Brim to get it flowing, I have experienced that often the blob will cause issues on subsequent loops of the brim or skirt. my solution is to use what's common called "Lead-in" on subtractive CNC cutting. where the tool head (nozzle) is offset from the start and comes in, its similar to a tail or perpendicular line

I would like to see the option to add a "Lead In Tail" or be able to set the first loop of a skirt of brim be a set distance away from the 2nd skirt loop, so that the blob doesn't interfere, reducing risk for print failure/spaghetti 

[image]

mtaumike, 2022-07-26:
Would this be on having `brim` enabled only? If so I think it's a neat idea but if not I don't want to be forced to have such happening on every print as I use a purge line which takes care of this for me as it is.

adamfilip, 2022-08-14:
Yes I think having it on the brim makes sense. optional toggle switch

closed 2026-09-29 by prusaslicer-bot[bot]