prusa3d/PrusaSlicer #8568: Arachne perimeter artifacts

dodod-00, 2022-07-27:
### Description of the bug

It is creating this weird artifact when using Arachne perimeter generator at higher layer thicknesses. I am using a 0.6mm nozzle and it does not happen below 0.2mm nor if using classic perimeter generator.
Having supports on or off makes no difference, just layer height.
Interestingly this part has a mirror image feature on the other side and this artifact does not occur there.

[https://www.facebook.com/photo/?fbid=10226460449431023&set=gm.3130757120518671&__cft__[0]=AZUuHuu85JGS8S653oL6fs5BUFIM550qwW4sDWZmmW9HKi8nsyJhOgqUMHQZMs9fEaah6wcz5AGZOiSF1A6_3Rp9iZR_Tuf2t0tIC913d5cwpSYU4dAz0DG-sVB7uWl6b5GL4OVyIO34gSawNpF0tg15QX9yCzxDAJbhoZ7G-5xC27_WTJvTJoDeWnplUtMJv74&__tn__=EH-R](url)

### Project file & How to reproduce

[https://www.facebook.com/photo/?fbid=10226460449431023&set=gm.3130757120518671&__cft__[0]=AZUuHuu85JGS8S653oL6fs5BUFIM550qwW4sDWZmmW9HKi8nsyJhOgqUMHQZMs9fEaah6wcz5AGZOiSF1A6_3Rp9iZR_Tuf2t0tIC913d5cwpSYU4dAz0DG-sVB7uWl6b5GL4OVyIO34gSawNpF0tg15QX9yCzxDAJbhoZ7G-5xC27_WTJvTJoDeWnplUtMJv74&__tn__=EH-R](url)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 Aplpha 3

### Operating system

Win 10

### Printer model

Prusa i3 MK3S+ 0.6mm nozzle

hejllukas, 2022-07-27:
Based on your screenshots, it seems that it could be similar to the #8446 issue, which is already fixed in the master_250 and will be part of the next alpha or beta.

Could you please upload your project file, so I can check if your issue is already fixed or if your issue is different from #8446?

dodod-00, 2022-07-28:
Thanks,

Unfortunately the project file is client confidential and I cannot share. But I will test it when the next alpha comes out.

 

Regards,

Neil 

Leave a review on  <https://g.page/r/CW48KYN-x8oREAI/review> Google

hejllukas, 2025-01-04:
We have fixed several similar issues some time ago. So we hope that this issue has already been fixed at 2.9.0.

closed 2025-01-04 by hejllukas