prusa3d/PrusaSlicer #8575: Slicing multiple build plates with custom G-code in between, or export to a folder of G-code's

3DCoded, 2022-07-27:
**Is your feature request related to a problem? Please describe.**
When I have a model with many parts that doesn't fit onto one bed, PrusaSlicer will arrange them into different plates. However, when I am done with the print, I have to move the next batch onto the bed and then re-slice.

**Describe the solution you'd like**
If there was a way to export G-gocde for multiple build plates at once with some sort of bed clearing or pause and beep G-code in between, to automatically start the next batch of prints, it would be very useful when printing MANY parts.

**Describe how it would work**
When clicking the Slice button, PrusaSlicer should detect that it has arranged multiple beds of parts and export multiple G-codes at once.

**Describe alternatives you've considered**
Manually combining the G-codes in a text editor.

viruha, 2023-03-08:
please consider multiple boards per project, like bambulab studio

MrTrick, 2025-05-17:
Perhaps this can be closed - I think what @3DCoded is asking for has been implemented. ( https://help.prusa3d.com/article/multiple-build-plates-on-prusaslicer_823894#g-code-preview )

3DCoded, 2025-05-17:
@MrTrick ,

Haven't been keeping up much with PS development. Thank you for letting me know!

closed 2025-05-17 by 3DCoded