prusa3d/PrusaSlicer #8578: Arachne does not overlap inner and outer perimeter, causes under extrusion - gap between walls

napter, 2022-07-27:
### Description of the bug

With a model where the wall width is 0.9 mm arachne is extruding two thin paths that are not the right thickness. The effect is that it leaves a gap between the walls as you can see in this photo. The left side is classic, the right is arachne.

To illustrate this, here are two models the curvy one has a 0.9 mm thickness. The cube is solid. We will print with wall thickness set to 0.49 mm, which is how I always print these models and it works great (more detail on this below). Top layers are turned off so you can see the cube walls. The two objects on the right are set to use arachne, the left is using classic.
[image]

Here it is sliced:
[image]

Notice the darker color on the arachne curvy object. That printed with thinner extrusion widths than the other three objects. The effect is that the thinner walls are separated by a small gap in the final object. This does not happen with classic, and as you can see the cubes were also fine.

**Here is what I believe is happening:** 
With classic perimeters, there is some overlap between the perimeters (see [prusaslicer documentation](https://help.prusa3d.com/article/layers-and-perimeters_1748)):
[image]

That is why when printing at a 0.49 extrusion width I have the model set to 0.9mm. In fact this is pretty much what prusaslicer itself recommends: 
[image]

Presumably when arachne determines the extrusion width for the cube with perimeters set to two it does overlap (I haven't actually tested this with a print but I believe this is the case). However, with the 0.9mm wall the two perimeters are an inner and outer perimeter - not two of the same perimeter. In this case Arachne is not selecting the proper extrusion width to overlap the layers and is extruding two thinner lines with no overlap leaving the gaps.

### Project file & How to reproduce

[arachne issue 2.zip](https://github.com/prusa3d/PrusaSlicer/files/9203083/arachne.issue.2.zip)

The project file above was used for the screenshots above. Reproduce by printing the file with arachne or merely looking at the gcode with view set to width. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5 alpha 3

### Operating system

Windows 10

### Printer model

Prusa MK3S+

crashberrypi, 2022-08-25:
This is also happening for 2.5.0-rc1 using Arachne. Does not happen with Classic mode.
[image]

hejllukas, 2022-09-01:
Hi,

I tried to print your attached 3MF on MK3S with an almost identical configuration, and this problem isn't there. The perimeters are so stuck together that I cannot separate them.
Same when I check the generated G-Code, the overlap between the two perimeters is there, and it is correct.
So, this isn't Arachne or the slicer issue. The issue is that your printer is under-extruding for some reason. Some issue with extrusion is also seen a little bit on the top surface infill in the first picture.

It should be mentioned that the Classic perimeter generator and the Arachne perimeter generator work differently with entered extrusion width. With Classic perimeter generator, whatever extrusion width you give it, it uses it and doesn't check if the object is over-extruded, that's up to the user to ensure. The slicer will only recommend to the user which wall thickness the object should have for entered extrusion width (shown below the perimeters input field).
In general, the Classic perimeter generator over-extrudes in a lot of cases. Because entered extrusion width will be used everywhere, even if it leads to over-extrusion.
Over-extrusion occurs especially in objects with only two internal perimeters or where perimeters from different object walls meet (for example, a hole in an object near another wall)
This means that the Classic perimeter generator, in many cases, can hide under-extrusion issues (caused by bad extruder calibration, wrong filament diameter, wrong extrusion multiplier, too fast printing, hotend unable to melt the specified amount of material, extrusion too wide for used nozzle diameter, extruder gears slipping, and much more).

On the other hand, Arachne doesn't over-extrude because it just takes the specified extrusion width as some initial value and calculates extrusion width according to the object wall thickness at every place. So that the object is filled with the exact amount of material, with no over-extrusion or under-extrusion caused by a slicer.
But this means that if a printer has any under-extrusion or over-extrusion issues, Arachne amplifies those issues compared to Classic, which will be your case.

If I look at the attached 3MF, you are using an extrusion width of 0.49 mm, and the slicer recommends that an object with two perimeters should have a wall thickness of 0.92mm. So, if you print an object with a wall thickness of 0.90mm, it should be over-extruded (if the printer extrudes the correct amount of material).
Conversely, when you use the extrusion width of 0.48mm, the slicer recommends that an object with two perimeters should have a wall thickness of 0.90mm. So, in your particular case, with the Classic perimeter generator, you should print an object with a wall thickness of 0.9mm with the extrusion width of 0.48mm, to not over-extrude the object. If you then compare the visualization of the extrusion width between Classic and Arachne with the 0.48mm extrusion width, there will be almost no differences.

So, I think if you use the Classic perimeter generator and the correct extrusion width for the object with a wall thickness of 0.9mm, with is 0.48mm in your case, you will have exactly the same issue as with Arachne.

You can also try to print an object with just one perimeter with the same speed (generally with the same settings) as in the attached 3MF with an extrusion width of 0.49mm and the Classic perimeter generator. And then, measure perimeter width using a micrometer or caliper (with sufficient measure accuracy), and you will see that width of this perimeter will not be 0.49mm, but smaller.

You can also contact our support about that.

closed 2022-09-01 by hejllukas

napter, 2022-09-01:
I included a link where it explains how perimeter overlap is necessary and
how it works. How does that perimeter overlap compare between arachne and
classic? If it is necessary it will still be necessary for arachne.

hejllukas, 2022-09-01:
You are right that overlapping perimeters are necessary for both Classic and Arachne. Both perimeters generator are using it.
The difference is that Arachne chose the extrusion width automatically. And based on computed extrusion width, it computes perimeter overlap. According to the G-code in your case, Arachne usually uses an extrusion width of 0.48mm. For the object with a wall thickness of 0.9mm, there is an overlap of 0.06mm (0.48 * 2 - 0.9), which is the same as for Classic, and an extrusion width of 0.48mm. And that should be sufficient for a 0.9mm wide object, and because it is sufficient, the slicer for 0.48mm extrusion width recommends the wall thickness of 0.9mm.

On the other hand, for the Classic perimeter generator, you can change extrusion width, which indirectly influences perimeter overlap. But the main reason for changing extrusion width (of perimeters) is dealing with the different wall thicknesses of different objects (that is what Arachne does automatically), and overlap should be the same for the same combination of layer height and nozzle diameter.
But there is also a possibility that changing of overlap (indirectly through extrusion width) is used to solve under-extrusions or over-extrusions issues. That is wrong in general. And if someone uses it like that (even without knowing that it is doing), then they could have issues with under-extrusion or over-extrusion when using the Arachne perimeter generator until they solve the real cause of the under-extrusion or over-extrusion 
Because Arachne expects that if a user enters some specific extrusion width, then the printer prints exactly this specific extrusion width.

detlefgerhardt, 2024-03-21:
It looks like I have the same problem here. with version 2.7.2. The outer walls delaminate completly, when Arachne is active. When I change to Classic or make the walls thicker (with infill) the problem is gone.

[image]