prusa3d/PrusaSlicer #8582: resume after pause dont heat the nozzle

soundpartner2, 2022-07-28:
### Description of the bug

So i do a lot of prints where i put nuts or ballberings into the print. i use prusa sliser, find the layer i need to make the break, rightclick in the layer line, and ad the m601 pause. 

This part works perfectly, BUT when the printer go to pause position it turn off the heat of the nozzle, and when i have placed my external part, and press resume print it start printing right away without reheating the nozzle first, and therefore causing problems.

when using this m601 pause it should automaticaly start reheating and shouldnt resume print untill after the nozzle have hit target temp. 

if it makes any diference i use octoprint and both printer and octoprint are newest versions

### Project file & How to reproduce

[ADX5D kasse1 v20.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9208081/ADX5D.kasse1.v20.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+x64

### Operating system

mac os 12,4

### Printer model

prusa i3mk3s+

neophyl, 2022-07-28:
Yes using Octoprint does make a massive difference.  That **is** the issue.  Please read the following thread in the MK3 firmware repositories -
https://github.com/prusa3d/Prusa-Firmware/issues/3148

soundpartner2, 2022-07-28:
what you are telling me is that this bug have existet and keep existing after 2 firmware updates, and i need to use 2 year old firmware and thereby not getting any of the good stuff addet the last 2 years? 

WOW thats bad

neophyl, 2022-07-29:
No what the thread is saying is that it is an octoprint configuration issue. When you set up octoprint correctly then it works.

stepikovo, 2022-07-30:
Can we close this issue then? :)

LegacyOfBob, 2022-09-05:
I'm curious as to why the behavior of M600 and M601 is so different when the function is similar. M600 via octoprint works fine and the printer remembers the printing temperature. Even if the nozzle cools there is a wait while it reheats and then prompts the user to remove the filament and reinsert. M601 the printer does not store the temperature and tries to cold extrude when resuming. The workaround mentioned in the octoprint forums seems to only mention manually pausing from the UI and not when calling a M601 at a certain layer in the GCODE.

rtyr, 2022-09-05:
This issue is not related to PrusaSlicer. I am closing this issue, you can continue discussion at https://github.com/prusa3d/Prusa-Firmware/issues/3148.

closed 2022-09-05 by rtyr