prusa3d/PrusaSlicer #8584: I hope to add a custom operation record button

xiaogui8dangjia, 2022-07-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I have many models that need a series of fixed operations, such as importing model thickness, cutting at a fixed height, simplifying, and then re exporting STL
I hope to customize a button to record a series of operations on the model
**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Similar to button editing in Rhino
**Additional context**
Add any other context or paint drawing about the feature request here.

dartrax, 2022-07-29:
I‘d rather like to be able to replace the model‘s STL with another one (updated or completly different), while the modifications (cutting etc. and, especially, paintings like seam painting) will be applied to the new STL.
In my opinion this has the same effect and no need for another Button on the UI.
I know that will not always work, like the surface with the painted seam may not be there with a new STL, but often it is and then the slicer should try to apply the painting to the surface that is most close to the old one.

dartrax, 2026-09-18:
This feature request is still valid and the underlying problem/use case still exists.