prusa3d/PrusaSlicer #8585: Show latest .stl / .3mf files that are created for import

Fliens, 2022-07-28:
**Is your feature request related to a problem? Please describe.**
When starting the slicer programm, it's very annoying to search for the file you just downloaded/created

**Describe the solution you'd like**
The slicer should present a list of the newest 3d-model files on the system. With that you could just pick the file you want or import it as before.

**Describe how it would work**
The slicer should search in some default dirs like Downloads/Desktop for the newest 3d-model files. It would also be nice if the user could specify some custom dirs. With that it would create a list of these {date:path} objects and sort them according to the date.

stepikovo, 2022-07-30:
Does actually any software do this?

foreachthing, 2022-07-30:
> Does actually any software do this?

I use [Everything](https://www.voidtools.com/)  for that. It indexes whatever and wherevrr you want

closed 2026-09-29 by prusaslicer-bot[bot]