prusa3d/PrusaSlicer #8586: Print head intersects with the print body at end of print

dannymulligan, 2022-07-28:
### Description of the bug

A recent version update of PrusaSlicer moves the print head to a position near the NW corner of the print bed, and much lower (close to the print bed) than before.  At the end of this print, moving the print head to this stop position caused it to intersect with the print, damaging the print.

Image included showing the damage to the printout.

I think I prefer the old behavior where the print head was moved up and away from the print bed at the end of the print.

### Project file & How to reproduce

[VoCore mounting-Wheel base tall.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9213511/VoCore.mounting-Wheel.base.tall.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2+x64

### Operating system

macOS 12.4

### Printer model

Original Prusa MINI+

neophyl, 2022-07-29:
Update your mini firmware. If you do a search of closed issues you will see this has been reported several times, here and in the mini firmware GitHub.

VegetableJoe, 2022-07-29:
I crosscheced travel path in slicer's g-code view and that looks ok. Can you update your firmware according @neophyl recomendation and let us know ? 
[image]

closed 2022-07-30 by stepikovo