prusa3d/PrusaSlicer #8587: Max Volumetric Speed

apcks, 2022-07-28:
### Description of the bug

Max Volumetric Speed doesn't work in version 2.5 alpha 3 for windows, you can change the value but it doesn't make any difference.

### Project file & How to reproduce

[Ver25aplha3.zip](https://github.com/prusa3d/PrusaSlicer/files/9213557/Ver25aplha3.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-alpha3+win64-202207151230.zip

### Operating system

Windows 11 Pro v22H2 Comp 22622.436

### Printer model

Creality Ender-3 V2

FidelCapo, 2022-08-08:
Can you please describe the problem more? What do you want to achieve? Also, the model in the 3mf files does not match the model in the photos. Thank you

apcks, 2022-08-11:
There are the same files as the photos, but the problem is the same, no matter the stl.
[PrusaProject.zip](https://github.com/prusa3d/PrusaSlicer/files/9311157/PrusaProject.zip)
As you can see in the photos in the original post, the two pieces are the same, but there is a difference in the printing. I could use other ways to solve this, but I tried this item, so it solved in Prusa 2.4.2, but when testing the same in Prusa 2.5 alpha 3 there was a big difference in the print of the part.

closed 2026-09-29 by prusaslicer-bot[bot]