prusa3d/PrusaSlicer #8588: Arachne generates disordered and discontinuous perimeter and hard to get smooth wall when printing

WeiTaoSJTU, 2022-07-29:
### Description of the bug

Disordered and discontinuous perimeters are generated for some model when using arachne, which cause rough surface and bed quality.
[image]
The outer most perimeter should be continuous and printed continuously.

### Project file & How to reproduce

Just load the stl and slicing by arachne.
STL file:
[Uploading sharp_detail.STL…]()
 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Prusa i3 MK3

fluksh, 2023-05-01:
Still the same issue in 2.5.2 and 2.6alpha6.
Adding arachne/classic as a (height-dependent?) modifier might let us use classic for the bottom layers where fractured lines are hell for adhesion, and arachne higher up, or reversed when necessary.

acekrystal, 2024-10-29:
Same issue here. I noticed my logo's and text does not come out nice as it used to do before.

Looking at it close, it seems to regularly try and make a few lines or dots on the outside, before starting the rest of the layer. Causing ooze to to get dumped, and sometimes floating dots and lines to be made in an area that you try to get as clean and nice looking as possible to keep the tekst and logo's nice...

Even tough I painted a seam line somewhere else on the print.. it keeps starting small dots in this area before doing the rest of the layer.
[image]

When switching to "Classic" This is nicely a single outer extrusion line giving a nice finish.
[image]

I tried multiple settings, but Arachne does not seem to bent to my will to print nice and clean outer perimeters.
While I do like Arachne for lots of things, it also seems to behave unpredictable for clean looking products

I would have loved if there was a way to force a (more) continuous outer perimeter. While keeping its dynamic solutions for everything inside the parts.

Here a close up of it starting a layer with this tiny dot inside the text area, causing ooze to also collect here after its layer change on the Prusa XL. After witch it then meets up with this dot later in the print.
[image]

System info : 
PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 27686), 64-bit edition
Total RAM size [MB]: 34,258MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 560.94)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1080/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
[... 335 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

closed 2026-09-29 by prusaslicer-bot[bot]