prusa3d/PrusaSlicer #8591: Prusa slicer auto-arrange fails when initial orientation to large for print bed

puterboy, 2022-07-29:
### Description of the bug

If I have two objects that together (without rotation) don't fit the print-bed as is, auto-arrange sends one of the objects far off the print bed (much further than the set minimal separation) - this itself is a bug.

But even worse, the 2 objects would both fit the bed if one of them were rotated 90 or 180 degrees.
However, the auto-arrange algorithm doesn't seem to attempt such rotation.

If I manually rotate one of the objects myself, then auto-arrange works.

So, it seems that somehow the auto-arrangement algorithm does not seek to do a rotation that extend beyond the area of the print bed even if doing so will allow a subsequent translation to make all the items fit.

### Project file & How to reproduce

As above.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa MK3S+

dartrax, 2022-07-31:
The doesn‘t rotate issue is a duplication of #5339.
The throw of build plate issue might be the same as #5653?

bubnikv, 2022-08-01:
If you click on the arrange button with a right mouse button, a dialog
opens up, where you may allow auto rotation. PrusaSlicer will then try to
rotate the objects.

puterboy, 2022-08-01:
> If you click on the arrange button with a right mouse button, a dialog
opens up, where you may allow auto rotation. PrusaSlicer will then try to
rotate the objects.

Thanks! Very helpful and that solves the rotation problem! (though wasn't obvious I could right click :)

Is there any way to make auto-rotation a default in the settings?

stepikovo, 2022-08-12:
> Is there any way to make auto-rotation a default in the settings?

When you select that option once it will then be the new default

puterboy, 2026-09-02:
Keep this alive until fixed!