prusa3d/PrusaSlicer #8592: Add ability to add notes and comments to a Prusa project

puterboy, 2022-07-29:
Some of my projects require customizations to the Prusa slicer project such as painting on supports or adding modifier shapes with different fill parameters etc.

These all need to be manually done every time I import a new version of the project from my modeling software (e.g., Fusion 360).
And unfortunately, sometimes I "forget" to make one of the mods necessary when I import a new version leading to wasted time and material.

It would be great if one could attach a comment or build notes to a Prusa project that one could reference within the project rather than having to create a separate unrelated word or text document for my notes.

This would also help when sharing Prusa project files with others since any details critical to printing or even final assembly could be added to the notes.

drf5n, 2022-10-01:
Yes.  There's `printer_notes`, `filament_notes`, and under Print Settings, there's `notes`.  Notes sometimes shows up as "Configuration notes" in this menu:

[image]

You can propagate the Print Setting notes from project to project, and they seem to be saved with the project when you save it, but if you save a project, and then edit the notes, the "File/Save Project" option is greyed out and you can't save your notes.  You can work around the potential loss of notes by making a more substantial change, and then saving the project.

It would be nice to have a separate `project_notes` field on the project page, distinct from the Print Settings `notes` that may get written over when switching print settings.

I'd use `project_notes` to keep track of where I got/store the source model file or the different settings I tried and their results.

dbarkelew, 2022-12-07:
If this helps. I keep my project notes with my printer settings. I discard my settings when I create a new project to clear them. When I change printer I have the notes checked and click transfer to keep them. They load back when I load old projects.

[image]

But I would prefer a dedicated project note section at some point, so for the printers that need to have notes across projects are separate.

puterboy, 2023-12-11:
Seems like such an obvious function to add -- allows the user to create notes on how to slice and/or print a given project.
Heck anybody posting a project to Thingaverse or Printables will typically add instructions to help others  properly slice and print so why wouldn't a user want to create similar in-situ notes to remind him/her on how to slice and print one's own projects...

Otherwise, I find myself taking notes in a separate text document which is klugy and prone to getting separated and lost from the 3mf file.

memsfactory, 2024-05-25:
I completely agree. A separate notes section that is independent of the profiles would be great. This way the text notes "travel" with the project but do not affect the printer profile info.

i-am-mike-davis, 2024-11-23:
> Seems like such an obvious function to add -- allows the user to create notes on how to slice and/or print a given project. Heck anybody posting a project to Thingaverse or Printables will typically [...]

I do the same thing. The information that the author posts alongside the STL and 3MF (and usually also the comments section too) is critical as it often adds context for understanding the project and also how to successfully print/modify. 

I usually copy/paste all that info into a README.md that I store alongside the stl or 3mf. I find that this is more useful to me than storing Notes inside the Print profiles. I want notes **about** the project, not about the print profile implementation.

Since my understanding is that 3MFs are zip archive files and people are reporting that they open the archive and manually insert a README.md file, it might not be too heavy of a lift for the slicer software to just read/write to a README.md file in the 3mf archive and display that information to the user in the GUI. Having not looked at the codebase however, this is pure conjecture on my part.

If somebody from the project could point me in the right direction, I'd be willing to take a swing at this in my free time.

guiweber, 2024-12-06:
This would be really useful to have!

metaxis, 2025-04-12:
I keep running into this need.  Project notes and per-object notes are a frequent enough use case that they deserve independent fields.  This avoids having to maintain a separate file and overloading random settings notes fields that can inadvertently leak into other projects.

With dedicated fields, these project and object notes can be presented in the sidebar, in the "Info" section and the per-object settings tree.

HonP, 2025-06-19:
I would use it too.

i-am-mike-davis, 2026-09-02:
Thank you for your attention to this matter.

puterboy, 2026-09-02:
Keep this alive until fixed!

Jan-Soustruznik, 2026-10-20:
Hello, A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16014.

closed 2026-10-20 by Jan-Soustruznik