prusa3d/PrusaSlicer #8593: Elephant Foot setting causing error in Arachne generator

kevinmichaeldudley, 2022-07-29:
### Description of the bug

If Elephant Foot Compensation is 0.1 or larger, the perimeters are not placed consistently in a symmetrical gear model.
If the Elephant Foot Compensation is 0.09 or smaller, perimeters are UNIFORM throughout the symmetrical gear model.

In the attached image, RED circles are around some examples of the inconsistency/bad areas, RELATIVE to the correct perimeters in the GREEN circles.

Additionally:  The concentric bottom FIRST layer has a few unexpected voids at the ends of lines.
In the attached image, those are circled in yellow.
[image]
[GearTeethVoids.zip](https://github.com/prusa3d/PrusaSlicer/files/9222539/GearTeethVoids.zip)

### Project file & How to reproduce

Open the project
Slice the project.
View the first layer.
Notice the printing of the TEETH profiles.
If Elephants Foot is 0.1 or larger there will be voids. RED CIRCLED
If Elephants Foot is 0.9 or less the TEETH profiles are uniform an properly printed. GREEN CIRCLED
[GearTeethVoids.zip](https://github.com/prusa3d/PrusaSlicer/files/9222553/GearTeethVoids.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Win 64 2.5.0 alpha

### Operating system

Windows 11

### Printer model

Mk3s

kevinmichaeldudley, 2022-09-04:
FYI: Tested and Not fixed in 2.5 Beta RC1

DragonFinlay, 2022-09-07:
I can reproduce this in 2.5.0-RC2 as well. A few points I noticed while testing this:

- It seems to be exclusive to the first layer, with elephant foot compensation enabled. Different compensation values give different results, but most of them are inconsistent and not how it should behave.

Example with elephant foot compensation set to 0.3 mm:

[image]

- The issue doesn't seem to be related to the geometry as the position of the "voids" is not related to the rotation of the object. With the given model, the voids always stay on approx. the left half.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]