prusa3d/PrusaSlicer #8594: Concentric infill TOP and BOTTOM leaving void area V2.5.2 and V2.5.0(Classic Perimeter Slice mode)

kevinmichaeldudley, 2022-07-29:
### Description of the bug

On this gear model, if using Concentric Top/Bottom infill, areas shown are not printed completely in the FIRST or LAST layer
[image]
.

### Project file & How to reproduce

Open the project.
Slice.
View FIRST/BOTTOM layer.
Voids as shown.
This is in version 2.4.2 AND 2.5.0(When in Classic Perimeter mode) as well.
[image]
[BottomLayerConcricFail.zip](https://github.com/prusa3d/PrusaSlicer/files/9222633/BottomLayerConcricFail.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 & 2.5.0 CLASSIC mode alpha

### Operating system

Windows 11

### Printer model

Mk3s

foreachthing, 2022-07-30:
You'd have to figure out the linewidth for those layers with concentric infill to make it tight... It has never worked otherwise - in my experience.

stepikovo, 2022-08-12:
This is a known issue of Concentric infill, you would have to manually change the extrusion width. Arachne, with dynamic extrusion widths, solves it.

I'm closing this as a duplicate of #3104 and many others.

closed 2022-08-12 by stepikovo