prusa3d/PrusaSlicer #8596: Where is the "add settings" button in v2.5.0??

3DCoded, 2022-07-30:
### Description of the bug

In v2.4.2, when I right click on an object, there is an "add settings" button that allows me to set many settings per-object, including brim, which is very important. In v2.5.0, there is no button.

### Project file & How to reproduce

Just in stall v2.5.0 and right click on an object. The "add settings" button is NOT there?!

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

v2.5.0-alpha3

### Operating system

macOS 12.5

### Printer model

Any printer on PrusaSlicer has this problem. I have tested the MK3S+ profile, my custom profile for my printer, and the MMU2S profile.

stepikovo, 2022-08-01:
Can you please include screenshots from 2.4.2 and 2.5.0 showing the issue?

3DCoded, 2022-08-01:
2.4.2:
[image]
2.5.0:
[image]

YuSanka, 2022-08-02:
I suppose, that you have Expert mode for 2.4 PrusaSlicer and Advanced mode for 2.5 PrusaSlicer

3DCoded, 2022-08-02:
Thank you! I switched to expert mode in 2.5 and it worked!!!!

closed 2022-08-02 by 3DCoded