prusa3d/PrusaSlicer #8598: Color Effects Slicing Process

Cookietyrant, 2022-07-30:
### Description of the bug

When changing the color of my STL from Black to White the slicer will ignore a bunch of the words and imagines on my file.... 

What I did was created one file with a Black background and white text which came out perfectly, but when I use the color change tool to inverse the colors making white my background and black my text the slicer seems to bug out during slicing.

The work around for this is to make the clear filament slot black and wont clear in teh software, but keeping the filament physically in the correct location.

### Project file & How to reproduce

What I did was created one file with a Black background and white text which came out perfectly, but when I use the color change tool to inverse the colors making white my background and black my text the slicer seems to bug out during slicing.

The work around for this is to make the clear filament slot black and wont clear in teh software, but keeping the filament physically in the correct location.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5 ALpha 3

### Operating system

Windows

### Printer model

MK3S+ MMU2S

Cookietyrant, 2022-07-30:
[image]
[image]
[image]

stepikovo, 2022-08-01:
Please provide us with the project file saved as 3MF

Cookietyrant, 2022-08-21:
Github is saying it is unable to accept 3MF files as an attachment for some reason.

Cookietyrant, 2022-08-21:
The issue seems to ne with using the 6mm nozzle vs 4mm

FidelCapo, 2022-08-22:
> Github is saying it is unable to accept 3MF files as an attachment for some reason.

3mf file must be zipped for attaching.

Cookietyrant, 2022-08-28:
[Defect STL.zip](https://github.com/prusa3d/PrusaSlicer/files/9440091/Defect.STL.zip)

closed 2026-09-29 by prusaslicer-bot[bot]