prusa3d/PrusaSlicer #8600: Aligned Seam not aligned even when forced in 2.5.0 alpha-3

5jvm0u4, 2022-07-30:
### Description of the bug

As the title described, the autoselected seam placement is not perfect to me, so I decided to force it to the place I desired, but failed.
Here are some screenshots of the situation and the 3mf file. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[Aligned seam not working in 2.5.0 alpha 3.zip](https://github.com/prusa3d/PrusaSlicer/files/9227156/Aligned.seam.not.working.in.2.5.0.alpha.3.zip)

### Project file & How to reproduce

Load the file and slice it.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 alpha-3

### Operating system

Windows 11

### Printer model

not relevant

kubispe1, 2022-09-07:
This issue is fixed in PrusaSlicer 2.5.0-rc2, so I am closing it. In case, let mi know.

closed 2022-09-07 by kubispe1

kubispe1, 2022-09-07:
[image]