prusa3d/PrusaSlicer #8603: PrusaSlicer build plate image going dark and missing the grid look.

barryk28, 2022-07-31:
### Description of the bug

Build plate in the slicer turns black shortly after opening the slicer in the desktop.
[image]

pictures attached. 

chat support asked to mention this here. 
happened out of the blue on 2.4 and 2.5 alpha

### Project file & How to reproduce

I can explain what happened to cause this

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 and 2.5 alpha

### Operating system

windows 10

### Printer model

MK3S+ with bondtech LGX shortcut mosquito FF

jmd73, 2022-08-01:
@barryk28  This seems like a duplicate of #8417

See comments there for a possible solution.

closed 2022-08-15 by stepikovo

hejllukas, 2022-08-19:
We tried to investigate this issue, and it seems that the two latest AMD GPU drivers, 22.8.1 and 22.7.1, probably contain some bug that is causing missing bed texture. For some reason, those new drivers have an issue when mipmaps are generated manually on CPU sides combined with texture compression.

We generate mipmaps manually on the CPU side because we previously had issues on some GPUs with generating mipmaps on the GPU side.
And for some reason, those new drivers stopped accepting texture mipmaps generated on the CPU side.

So we made a workaround that for AMD GPUs with drivers newer than 22.6.1 (the latest working driver), we generate mipmaps on the GPU side, which seems to work correctly now.
This workaround will be part of the upcoming PrusaSlicer 2.5.0-rc1, so the texture of the bed should start showing again.

For older PrusaSlicer versions, the only way (at the moment) to get texture showing is to downgrade the AMD driver to 22.6.1 or 22.5.1 (WHQL).

bubnikv, 2022-08-19:
Please try with PrusaSlicer 2.5.0-rc1. The issue should be resolved.