prusa3d/PrusaSlicer #8604: inconsistent finish/reflectivity on vase

pntucker2, 2022-07-31:
### Description of the bug

I expected a consistent shiny finish on the walls - which I got in the production version -gold is old, copper is alpha 3
[image]
[image]
[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/9229531/Archive.zip)

### Project file & How to reproduce

File looks fine in the slicer but the results are not. One of the copper vases printed with the strange gap as well as the change in reflectivity. 

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0-alpha3+arm64-202207151229

### Operating system

PrusaSlicer-2.5.0-alpha3+arm64-202207151229

### Printer model

MK3S

catmonger, 2022-08-02:
Could we have any extra details that someone could use to fix this bug. For example a gcode file, stl/obj etc, configuration, device  even though it is most likely some sort of Distro by the cpu architecture.

pntucker2, 2022-08-02:
I had attached a couple of stl  files and a gcode to my original post - I thought.

kubispe1, 2022-08-03:
Thanks for your report. We didn't find any issue of slicing due to PrusaSlicer-2.5.0-alpha3. Still we wanted to be sure there is no hidden trouble, so we tried to print your vase. I used your. 3mf and printer MK3S. Here I attached the result. Seems to be fine and also with some shinny look :D
[image]

@pntucker2 can you check your printer hardware, especially condition of your nozzle or hot-end. Have a nice day.

pntucker2, 2022-10-11:
I did some tests and can confirm it’s Prusaslicer alpha 3 problem. Attached are screen shoots of the view of a vase in Prusaslicer and the exact same file (I saved it as .3mf in Prusa ) opened Cura 5.1 - note the stringing in the interior which corresponds to the different surface on the exterior.
Peter Tucker
5955 Balsam St. #305
Vancouver  V6M 0A1

DragonFinlay, 2022-10-12:
Sorry, I don't see screenshots attached. :)

closed 2026-09-29 by prusaslicer-bot[bot]