prusa3d/PrusaSlicer #8608: Extruder exclusion zones

gnif, 2022-08-02:
It would be handy to be able to define an exclusion zone that the print head is not allowed to enter, this way it would be possible to define a region to avoid so that a part can be inserted into the print part way through the process, allowing printing objects/components in place.

**Describe how it would work**

Importing a shape and defining it as an "exclusion shape" would likely work best, this way one could define an area to avoid crossing. For example, the orange transparent area would be define as a "exclusion zone" so that a part could be dropped into the cavity part way through the print:

[image]

dartrax, 2022-08-02:
So by saying „print head“ you must mean not only the nozzle but the whole print head including the gantry.
This would require that one can import an STL that represents the print head and gantry, so the slicer can decide how to travel around the protruding part or even show which areas can‘t be reached anymore.

In general, I‘d like to see the option to define the print head and gantry. Mainly because that would enable prusaslicer to arrange multiple objects without collisions with the print head and gantry when printing sequentially (one object after another). And your idea would be another logical use case, than!

gnif, 2022-08-02:
> So by saying „print head“ you must mean not only the nozzle but the whole print head including the gantry.

No, just the hot end and nozzle geometry.

dartrax, 2022-08-02:
But if you only define hotend/nozzle the print head will crash into the protruding part. For example with the part cooling duct 😉 
It will depend on the part geometry and printer (how many space between gantry and nozzle) if the simpler approach (like defining a cylinder diameter and height) will work.

gnif, 2022-08-02:
Yes I understand, however it seems my use case differs from what you are thinking. In my example just a few millimetres of clearance is all that is needed. Ie, in the example above while the border is being printed, prevent the nozzle from travelling across the excluded area, but rather around it. This way one could pause the print and drop a part in before it completes.

dartrax, 2022-08-02:
I think it won't take too long until other users try to insert higher protruding parts ;-)
For your case, isn't the "Avoid crossing perimeters" option enough?

gnif, 2022-08-02:
It might be, I had not thought of using that feature for this.

MendyBerger, 2023-07-13:
Think this can also help with stringing. If there's one spot where stringing is not acceptable, you could make that a exclusion zone.

closed 2026-09-29 by prusaslicer-bot[bot]