prusa3d/PrusaSlicer #8617: Infill density vs. pattern

espr14, 2022-08-03:
### Description of the bug

By switching the infill type the final weight of the model changes a lot. This is not caused by the infill connections and anchoring. It happens even in big inner volumes.

### Project file & How to reproduce

In the attached project you can see that 15% grid infill weighs 4.3 g but honeycomb weighs 5.43 g. Moreover, the 100% infill weighs 27.5 g so the 15% infill should take 4.125 grams only.
I know that it's hard to get microgram precision for all infill types but it's easy to create a calibration matrix that minimizes the error for large (infinite) objects. The calibration curve can be non-linear but you can approximate it by few points and interpolation.

[Shape-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9250581/Shape-Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 10

### Printer model

Prusa MINI

bubnikv, 2022-08-03:
Thank you. Please provide a pull request.

espr14, 2022-08-07:
Calibration coefficients for densities:
1 5 10 20 40 60 80 99%

recti	0,878387097	0,99379562	0,986594203	0,99379562	0,997435897	0,998655257	0,999724644	0,999914688
grid	0,9725	0,99379562	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
trian	1,008518519	0,99379562	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
stars	1,008518519	0,99379562	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
cubic	1,008518519	0,99379562	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
line	0,938965517	1,008518519	0,990181818	0,995612431	0,997435897	0,998655257	0,998807886	0,999173462
conce	0,756388889	0,845652174	0,861708861	0,868580542	0,87136	0,87275641	0,873106212	0,873548283

honey
3d honey	0,611397059	0,719290657	0,727471566	0,742079429	0,745405648	0,741858736	0,745907154	0,742611637
gyroid	0,955747126	1,071520619	1,049873737	1,061941252	1,066025641	1,09383907	1,093719171	1,096118509

hilbert	1,008518519	1,001102941	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
archimed	1,008518519	1,001102941	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
octagram	1,008518519	1,001102941	1,001102941	1,001102941	1,001102941	1,001716738	1,001563218	1,001400446
adaptive	1,433157895	1,3615	1,34137931	1,351364764	1,354726368	1,356976744	1,357258567	1,356021127
support	1,433157895	1,3615	1,34137931	1,351364764	1,354726368	1,356976744	1,357258567	1,356021127
lightning	1,701875	1,448404255	1,433157895	1,440740741	1,448404255	1,532645403	?	?

espr14, 2022-08-10:
> Thank you. Please provide a pull request.

https://github.com/prusa3d/PrusaSlicer/pull/8659

espr14, 2022-08-10:
Hello @bubnikv. Can we get this https://github.com/prusa3d/PrusaSlicer/pull/8659 into 2.5.0 version? The main reason is that the new lightning infill is cheating a lot. It generates lower density by 33% (mostly) which is a record among all infills. We can split discussing about implementation from discussion about the final calibration data. Thank you.

espr14, 2022-08-10:
Afterwards, it would be nice to update this document about infills:
https://help.prusa3d.com/article/infill-patterns_177130

gringer, 2022-12-15:
pr #6434 includes fixes to the 3D honeycomb pattern to improve the density accuracy (among many other improvements). It has an exact calculation for low densities, with a linear scale factor of 1.1 applied above 42% infill based on the filament length reported in PrusaSlicer.

[42% infill is where the pattern switches from a more faithful representation of the truncated octahedron honeycomb to an elongated 2-layer repeating pattern, for practical / mechanical purposes].

gringer, 2026-09-03:
This is still an issue in PrusaSlicer because there is no density adjustment in the 3D Honeycomb infill code; all it cares about is the oddness of the Z layer ID:

https://github.com/prusa3d/PrusaSlicer/blob/6f510128d7c2e543b62919b74bea7e876f564205/src/libslic3r/src/libslic3r/Fill/Fill3DHoneycomb.cpp#L172

This issue has been fixed in OrcaSlicer. See the `Fill3DHoneycomb.cpp` source for more details:

https://github.com/OrcaSlicer/OrcaSlicer/blob/main/src/libslic3r/Fill/Fill3DHoneycomb.cpp