prusa3d/PrusaSlicer #8618: Unable to split (because outside print area?)

reedy, 2022-08-03:
### Description of the bug

Trying to print `flaps.stl` from the attached zip on a Prusa Mini. It seems because it is "outside the build area", it won't let me split it (which would allow it to be in the build area).

### Project file & How to reproduce

[Uploading split-flap-counter-fully-printable-model_files.zip…]()

As per the instruction on https://www.printables.com/model/69603-split-flap-counter-fully-printable

>I recommend splitting the flaps.stl in PrusaSlicer to fit smaller build plates.
>Right-click the object -> Split -> To Objects.

So set the printer to Prusa Mini, import `flaps.stl` from the zip, rigth click (well, 2 fingers on a mac trackpad), and the split option is greyed out.

[image]
[image]

So I am unable to split them to make it fit on the build plate (by re-arranging or only just printing some of the flaps).

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+x64, Version 2.5.0-alpha3+x64

### Operating system

macOS 12.5

### Printer model

Prusa Mini, 4.3.4 with runout sensor

jmd73, 2022-08-03:
Just for information it seems to work on windows, at least with 2.5.0 alpha 3.

@reedy 
Two workarounds that should work:
1: Rotate the object 45 degrees and it should fit on the build plate.
Then you can arrange the objects and use "place on face" to put the flat side down on the build plate.

2: Change printer to a bigger one (in PrusaSlicer), for example an MK3S+ (if you need to you can arrange the part so it is in the middle of the platter).
Split to objects, and change back to the mini.

reedy, 2022-08-03:
Thanks, yeah. I'd thought about the latter as a potential workaround...

On 2.5.0-alpha3+x64 I tried rotating it, so it does fit, and it doesn't seem to enable the split feature, unfortunately.

[image]

[image]

So if it works on Windows, is it something weird on the macOS build?

Which I guess mean I should try the bigger printer one and see if that "fixes" it....

reedy, 2022-08-03:
[image]

Yeah, it seems it's just broken on macOS...

jmd73, 2022-08-03:
Here you have a zip file with a  project for the Mini (.3mf), where I have split it to object if you don't succeed:
[flaps.zip](https://github.com/prusa3d/PrusaSlicer/files/9254260/flaps.zip)

reedy, 2022-08-03:
That's really helpful, thanks. It was going to be my next request (probably on https://www.printables.com/model/69603-split-flap-counter-fully-printable ) to do that so I could at least be unblocked.

I'll have a play on a Windows machine "soon", and see whether it works there for me too. Which probably helps narrow down if I'm doing something stupid, or if it's just unintuitive (I'm sure I've used it fine before on other models... Possibly on older version of PrusaSlicer), or indeed just broken on macOS for some reason.

reedy, 2022-08-03:
I note it's done some... weirdness with the alignment on a couple (highlighted and the one on the most right). And does it again if I move them around... But that's another issue/bug :)

[image]

jmd73, 2022-08-03:
@reedy It depends on how the object is rotated. If you rotate all objects so they are the same way, they will align as expected.

jmd73, 2022-08-03:
@reedy What mode are you in?
After looking at the code, it seems you have to be in advanced or expert mode.

[image]

reedy, 2022-08-03:
> @reedy What mode are you in? After looking at the code, it seems you have to be in advanced or expert mode.
> 
> [image]

Ah. Ffs. Yeah, that explains it. But that's not very intuitive :/

stepikovo, 2022-08-15:
@reedy You can also always use the top toolbar, where Split to objects is available even in Simple mode.
Can we close this issue?

reedy, 2022-08-15:
> @reedy You can also always use the top toolbar, where Split to objects is available even in Simple mode.

How does that make any sense? Why have one guarded by mode, and the other not? If there's one in the big menu that works, why not have the "right click" menu one work too under the same circumstances?

Then the lack of any sort of tooltip on the right click menu to give a hint...

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena

reedy, 2023-02-01:
Sweet, thanks!