prusa3d/PrusaSlicer #8620: Function 'int(layer_z)' fails to round properly for some seemingly round (whole) numbers of layer_z in GCode

puterboy, 2022-08-04:
### Description of the bug

For certain whole numbers of layer_z, the expression {int(layer_z)} equals (layer_z-1) rather than layer_z.

Specifically for some (seemingly) whole number values of layer_z
1. int(layer_z) = layer_z-1
2. int(layer_z+.000000000001) = layer_z
3. printing layer_z or even layer_z*1000000000 doesn't show any fractional component of layer_z

### Project file & How to reproduce

Create any model with layer height 0.2
Add the following two lines to the section "Before layer change G-code":

{if int(layer_z) != layer_z && int(layer_z+.01) == layer_z}; T FailedRounding1: {layer_z} {int(layer_z)}{layer_z*1000000000}{endif}
{if int(layer_z+.0000000000001) != layer_z && int(layer_z+.01) == layer_z}; T FailedRounding2: {layer_z} {int(layer_z)} {layer_z*1000000000}{endif}
{if int(layer_z+.00000000000001) != layer_z && int(layer_z+.01) == layer_z}; T FailedRounding3: {layer_z} {int(layer_z)} {layer_z*1000000000}{endif}

Then grep the G-code for 'FailedRounding'...

On a model with height of 91.4, I got the following when grepping the file, setting aside the mention of the code itself,
I go the following counts:

FailedRounding1 = 29
FailedRounding2 = 0
FailedRounding3 = 25

So, it seems that the internal representation of layer_z differs from the expected (and printed) representation of layer_z by a small error of ~1e-12 that messes up expected integer counts.

Recommended fix would be to round layer_z to say 1e-6 which is way smaller than any real-world slice but larger than the error introduced above.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa MK3S+

gudvinr, 2022-12-21:
I experienced same issue when I tried to use macro to do something every 5mm like this:

```
{if layer_z/5 == int(layer_z/5)}do something{endif}
```

If I print out `{layer_z/5}` and `{int(layer_z/5)}`, first will be 1 and second is 0.

puterboy, 2026-09-02:
Keep this alive until fixed!