prusa3d/PrusaSlicer #8622: Holes in honeycomb infill and smart infill request

jzagaja, 2022-08-04:
Please add holes to the honeycomb infill so that air can penetrate cells, see:
https://www.youtube.com/watch?v=093UQ_6GK1M

Please add smart infill like this one in Chisel slicer plus holes between cells like above:
https://www.instagram.com/tv/Cb0N9IzvQH1/?utm_source=ig_web_copy_link
this will spead up primts

stepikovo, 2022-08-18:
Please don't combine 2 or more issues/feature requests together.

If you need an open infill, use gyroid, 3D honeycomb or any other interlaced infill. Honeycomb infill is meant to be closed. Will this solve your issue?

For the new infill please open a new ticket.

jzagaja, 2022-08-18:
When watching 10% 3D honeycomb I do not see air can easily penetrate cells. Can you illustrate?

stepikovo, 2022-08-18:
[image]
If this is not enough, please select Gyroid.

jzagaja, 2022-08-18:
Ideally one need large cells for large objects.

stepikovo, 2022-08-25:
Air can easily get through the mentioned infills. Can we close this issue?

jzagaja, 2022-08-25:
Will test soon, keep open.

gringer, 2022-12-15:
The 3D Honeycomb infill in PrusaSlicer is broken; it was an early experimental code that was pulled from Slic3r just before it was updated, and hasn't been fixed. Please don't recommend it for use.

I've had an open pull request (#6434) to get this fixed in PrusaSlicer, which has been sitting untouched for over a year. The fix substantially increases the mechanical stability, reliability, and infill percent accuracy, and should actually be more airy because it doesn't switch directions as often.

gringer, 2026-09-03:
This is still an issue in PrusaSlicer because all it cares about is the oddness of the Z layer ID:

https://github.com/prusa3d/PrusaSlicer/blob/6f510128d7c2e543b62919b74bea7e876f564205/src/libslic3r/src/libslic3r/Fill/Fill3DHoneycomb.cpp#L172

This issue has been fixed in OrcaSlicer. See the `Fill3DHoneycomb.cpp` source for more details:

https://github.com/OrcaSlicer/OrcaSlicer/blob/main/src/libslic3r/Fill/Fill3DHoneycomb.cpp