prusa3d/PrusaSlicer #8623: Unable to enforce support blockers

jonwhunt, 2022-08-04:
### Description of the bug

Using PrusaSlicer 2.4.2
Windows 10

I've sliced this model before in PrusaSlicer few days ago.  Sliced it today and it placed support up the back side that I can't use enforcers to block.  It looks like slicer wants to suppose the 1mm where the support in that top hole sticks over edge of model.  I've set support to snug, changed support angle and spacing, changed from Supports Only From Bed to other options, tried changing paint on supports, I tried placing support blockers all the way up the back to the area near top of cut out, nothing helps.  The tiny bit of support up this back side isn't actually supporting anything, its just going up and merging with the support I painted into cut out area.  I want to print in this orientation to cut down on amount of support needed on other side.  I've managed to print this model once before using PrusaSlicer where it didn't add that tiny bit of support up the back.... not sure why its adding it now.

[image]

### Project file & How to reproduce

[Cord Plug Back slicer project.zip](https://github.com/prusa3d/PrusaSlicer/files/9258461/Cord.Plug.Back.slicer.project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

CR10s

neophyl, 2022-08-04:
There's always a little bit of expansion beyond the area you define, just check out your dark green bottom layers to see what I mean.  You just need to trim back the enforced area slightly.  Have a look at the painted area in the attached as an example of the way you need to tweak things sometimes.  
[Cord Plug Back slicer project_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/9258733/Cord.Plug.Back.slicer.project_neo.zip)

Your project has more problems than the support though.  The structure you have added using box objects has 3 overlapping and interpenetrating objects.  This means PS is going to lay down multiple sets of plastic into the same area as they aren't merged into one object.  All separate objects are sliced independently, if you as the user don't check for such overlaps then things can get messy.
Also as they aren't part of the main object then the support for the main object goes right through them, leading to more printing where PS is going to try putting 2x the plastic into the same volume.  The attached project also has those objects merged to stop that happening.

stepikovo, 2022-08-18:
@jonwhunt you can decrease the _Closing radius_, even 1.99 mm should help. 
Would this solve your issue?

closed 2026-09-29 by prusaslicer-bot[bot]