prusa3d/PrusaSlicer #8626: Don't emit start_filament_gcode if it is included in start_gcode

gudvinr, 2022-08-04:
**Is your feature request related to a problem? Please describe.**

I use custom gcode commands per-filament to set up things like linear advance coefficient, PID values for bed and hotend temps, etc.  
Those parameters cannot be set through UI so only way to add them is to use `start_filament_gcode`.

But, for example, PID values should be set before you start heating which happens inside `start_gcode`.   
You can add `[start_filament_gcode]` and PS will correctly insert its content but it will still append `start_filament_gcode` after set up commands after `start_gcode`.

**Describe the solution you'd like**

I'd like to be able to insert `[start_filament_gcode]` in `start_gcode` without running it twice in same file.  
Same could be done for `end_filament_gcode` and `end_gcode` but that doesn't bother me personally.

stepikovo, 2022-08-18:
We are already considering adding PID values for each filament, but it will take some time to implement it.

gudvinr, 2022-08-18:
Thing is that newest marlin release has experimental [MPCTEMP](https://marlinfw.org/docs/features/model_predictive_control.html) algorithm for hotend temp while bed still uses PID.  
MPC uses different approach and different set of parameters, thus it makes PID values basically useless and example I used still remains valid.

BloodyRain2k, 2023-03-11:
> We are already considering adding PID values for each filament, but it will take some time to implement it.

Would it be easier to implement a flag that could be checked instead? Something like `{if pre_start}<pid_gcode>{endif}`.
Then it could be used as described for the beginning of the print before the printer starts warming up, but wouldn't interfere with later references.

closed 2026-09-29 by prusaslicer-bot[bot]