prusa3d/PrusaSlicer #8628: Extra print lines outside model in PrusaSlicer 2.4.2

Gefionious, 2022-08-04:
### Description of the bug

When printing a simple model (of a cookie cutter) I am getting extra lines not in the model nor does it show up in any gcode viewer that I tried (4 or 5) but yet they remain. Using PrusaSlicer latest (2.4.2),prusa i3 mk3s+, octoprint (latest). Printing in PLA.

Two photos show the troubled print. "Troubled Print"
Photo shows the model in PrusaSlicer: "Screenshot"

To eliminate variables in my setup, I tried to redo the effort in Simplify3d slicer (with no issues):
Photo shows the model printing from Simplify3D: "Clean Print"

I also uploaded both gcode files, the PrusaSlicer project file.

Please let me know if there is anything else I can provide.

Thank you!

### Project file & How to reproduce

[PrusaSlicer 2.4.2 Extra Lines.zip](https://github.com/prusa3d/PrusaSlicer/files/9264079/PrusaSlicer.2.4.2.Extra.Lines.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Edition	Windows 10 Enterprise Version	21H2 Installed on	‎7/‎22/‎2020 OS build	19044.1826 Experience	Windows Feature Experience Pack 120.2212.4180.0

### Printer model

Prusa i3 mk3s+

kubispe1, 2022-08-11:
Thank you for your report. I observed it quite closely, but I can not find the reason, why you got this ugly output in your case.
At the beginning, I expected that it is some stinging on a travel path. So, there is a problem with some settings (temperature, retraction and so on) 
[image]

But then I printed from your 3mf (PS 2.4.2), but results looks quite ok. There are some defects but still…
[image]

Only thing I could recommend to you (if you want :D ), is to use updated slicer version PrusaSlicer 2.5.0-alpha3. It improve especially thin walls of your cutter. The reason is Arachne that is implemented in this release.
[image]

rtyr, 2022-08-11:
Could you please try print it again directly from SDcard? (without Octoprint).

What is your firmware version?

Gefionious, 2022-08-15:
My apologies for the delay in responding. My firmware is 3.11.0-4955. I will try printing from the SD card today (it will take me a while to find the correct gcode file, lol, I tried about 30 different slices. Stay tuned!

Gefionious, 2022-08-16:
OK, I was finally able to get this reproduced. No difference in printing with octoprint or SD card. I am beginning to believe that there might be some weird combination of settings going on. When I start over with fresh profiles, I could not get it to replicate, but when I went back to my project file save I could. This might be very difficult to chase down and with the workaround being not using my old profile any more, maybe that is good enough, especially with the new slicer algorithm coming?
[image]
[image]

kubispe1, 2022-08-16:
Thanks a lot for your tests and I am glad to hear that default profile helps you to get better results. I will let this issue open for a while maybe someone else can see what's the problem.

rtyr, 2022-08-16:
I believe I know what happened. I can see you have "External perimeters first" parameter enabled, which caused that some perimeters were printed in the air. There is some overhang in that area which doesn't allow using such setting.

[image]

I am closing this issue.

closed 2022-08-16 by rtyr

Gefionious, 2022-08-17:
Thank you for helping me understand the issue!

Gefionious, 2022-08-17:
UPDATE! I tried the same model with 2.5 beta1 with almost default profile and it came out amazing! My best print in a thin wall situation ever. Keep up the good work! Very excited, apologies for all the exclamation points. :)

AkashicSeer, 2023-02-13:
prusa is giving me "a toolpath outside the print area was detected." error. For some stupid reason, prusa is adding these two wild lines for no reason. I am using version 2.5.0  This is just directly importing it, no changes to any settings etc. Just default everything. It does it with STL and OBJ files. I created this 3d printer power box with fusion 360. I have never had any extra lines before. Since this works in Cura just fine, this can only be a Prusa issue. Any idea what would cause this? Check the angle too. It is seriously odd, I had nothing in the drawing that would make this wild angle. Before slicing it looks right. After slicing it has the added lines at an angle.

I only started using Prusa because Cura couldn't even slice my last simple double hollow cylinder soldering smoke filter design, at all. Now prusa can't slice this correctly. Unfortunately most of my designs are complex and take 24+ hours to print. LOL 

[image]

kubispe1, 2023-02-13:
@AkashicSeer, please attach 3mf. I think that this is a different kind of issue.

AkashicSeer, 2023-02-14:
I have no idea what a 3mf is, sorry. Here is an STL file

kubispe1, 2023-02-14:
File/Save Project as there it allow you to save .3mf
- Stl is not attached

rtyr, 2023-02-14:
You have to use web interface, attaching files via e-mail is not possible.

AkashicSeer, 2023-02-17:
Thank you! Sorry it took me a few days. Here is the file. I hope this helps
someone working on Prusa figure out why this even happened? This is the
only thing I have had it happen with. And this file works with Cura, no
extra lines

AkashicSeer, 2023-02-17:
Sorry I have no idea how to upload a file to this POS. It won't show the 3mf files so I can't upload it here. I tried via email. I mean github is a site for code why the hell doesn't it allow you to upload and share a F***ing code file? Not like Microsoft doesn't have the money to hire programmers to add code to properly filter viruses etc. 

[image]

AkashicSeer, 2023-02-17:
Hopefully this is useful to someone working on the core of Prusa. I like prusa, but it couldn't slice this without adding extra lines like the picture above. Cura slices this one fine and quickly. However Cura couldn't slice my last project a special charcoal filter for soldering I made that was a large cylinder to hold the charcoal and another cylinder on top holding the fan. It simply just failed. LOL

[printer-power-box v57.zip](https://github.com/prusa3d/PrusaSlicer/files/10763143/printer-power-box.v57.zip)

Gefionious, 2023-02-17:
@AkashicSeer  check this setting out, this is what went wrong on my original post:

From my issue above... 

> _"I can see you have "External perimeters first" parameter enabled, which caused that some perimeters were printed in the air. There is some overhang in that area which doesn't allow using such setting"_

Based on past troubleshooting, try and use all of the default settings and see if that improves things, then change one setting at a time until it breaks. There may be a bug but it takes a lot of work to troubleshoot these kinds of things, so it is better to narrow the issue to a particular setting/feature before submitting it to a bug report.

Good luck!

AkashicSeer, 2023-02-17:
Thanks, I'll check it out. I'm using the latest Prusa 5.2 version. When you
start a new project it sets the defaults back. Even with that unchecked and
all defaults it still makes the extra lines. I doesn't do it with other
projects though.