prusa3d/PrusaSlicer #8630: Extreme Over Extrusion on external perimeters when "Detect bridging perimeters" is enabled 2.5.0 Alpha

vertigo235, 2022-08-05:
### Description of the bug

External bridge perimeters are over extruded by crazy amount when Detect bridcing perimeters is enababled in the 2.5.0 alpha 2 and 3. 

[image]

### Project file & How to reproduce

Load item that uses bridging perimeters, turn the feature on and observe. 

Seems easily reproducible, if you can't, I'll share a project.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 Alpha 3

### Operating system

Win 10

### Printer model

Prusa machine running klipper firmware

rtyr, 2022-08-05:
It is caused by your extreme bridge flow value. Changing it to some sensible value should fix the issue (default is 1). My guess is that original 3MF was created in SuperSlicer, which uses different values for bridge flow (for example 80%, instead 0.8).

[image]

Closing.

closed 2022-08-05 by rtyr

vertigo235, 2022-08-05:
Interesting, thanks for your help.  I thought that I had accounted for this by using an included profile but maybe those did not update with the Prusa profiles, something else I need to look into. 

Hopefully this issue will help others if they search.