prusa3d/PrusaSlicer #8631: Printing rollers / cylinders

Ret-Marut, 2022-08-05:
When printing rollers or cylinders, there is always a joint on the same place in every level (height), which makes the roller not running smooth.

Could it be possible to detect cylinders / circles in STL files and turn them some angle with every layer – maybe a fraction of the roller height.

If a detection is not visible, could there be some sort of support for doing that on the generated gcode? In the sliced preview the joints are clearly visible.

DerekSauer, 2022-08-05:
That's the seam that forms where the extrusion line starts and ends and is one of the trademark indication of an FDM print. If you change the seam position to "random" and reslice the part you should get the behavior you desire.
[](
https://help.prusa3d.com/article/seam-position_151069)

Ret-Marut, 2022-08-06:
Ah, super! Thank you. I should have RTFM ;-)

closed 2022-08-06 by Ret-Marut