prusa3d/PrusaSlicer #8632: Prussa bed issue in PSlicer

huzargarden, 2022-08-05:
### Description of the bug

HI, just after updating the Prusa slicer to version 2.4.2, I've lost the picture of the bed. Is now just a black shape. 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 11

### Printer model

Prusa Mks3+

neophyl, 2022-08-05:
Duplicate, been reported several times already, for example https://github.com/prusa3d/PrusaSlicer/issues/8508

rtyr, 2022-08-05:
I am closing this as a duplicate. It seems to be related to some AMD driver update.

closed 2022-08-05 by rtyr