prusa3d/PrusaSlicer #8633: Arachne bulges

meermanr, 2022-08-05:
### Description of the bug

When slicing a cut-away of a tubular model there are extrusions significantly beyond the bounds of the original model.

This does not occur when using classic perimeter generation. I suspect this exposes a corner case.

### Project file & How to reproduce

### Screenshots

Original model before slicing. Note the flatness of the cutaway wall:

[image]

Sliced output using Arachne:

[image]
[image]

Resliced with Classic perimeter generation for comparison:

[image]

### Project file

Project file: [Pressure Washer-HighPressureNozzle_cover2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9269652/Pressure.Washer-HighPressureNozzle_cover2.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0-alpha2+x64

### Operating system

macOS Monterey Version 12.4 (Intel)

### Printer model

Original Prusa MINI+ with 0.6mm nozzle

meermanr, 2022-08-05:
Retested in Version 2.5.0-alpha3+x64, no change versus alpha2.

hejllukas, 2022-08-08:
I wasn't able to reproduce that on your model (STL). Could you please upload 3MF project file (File->Save Project as...)?

closed 2026-09-29 by prusaslicer-bot[bot]