prusa3d/PrusaSlicer #8636: Export ALL individual STLs from a platter.

WhiteDragonD, 2022-08-05:
I have 100+ files, separated into groups of 20 (5 folders).  I open Prusa Slicer 2.4.0 and drag 20 files into the app.  Select All, Rotate 90 degrees (or click on Face sides that I want on the plate).

Problem is, I have to right click each one and Export as STL then save/overwrite.

I want to be able to select all the STLs, Right Click, and Export all.  This can be to replace the ones on the hard drive or even to save to a different folder if you can't auto replace.

I tried to create a keyboard macro.....  but I didn't see a keyboard key for "Export as STL..." file.  If there was a short cut, I could at least create a macro that..  Export as STL -> Save -> Overwrite (yes) -> Down Arrow (to select next one)..... then repeat the process.

FidelCapo, 2022-08-08:
I think you can use the Export Plate as STL... function (File -> Export -> Export Plate as STL... Is this function what you are looking for?

WhiteDragonD, 2022-08-08:
> I think you can use the Export Plate as STL... function (File -> Export -> Export Plate as STL... Is this function what you are looking for?

No, I need to export each file individually so I can overwrite the original files currently in a folder.

MNeMoNiCuZ, 2022-08-17:
+100

This feature would be great to have.
It's not unreasonable to have a lot of different items, and you wanting to export them individually.

The individual items should maintain the name of the object in the list, i.e. like it's doing currently, but it only supports one item.

Note that the ability to export all objects as one mesh still needs to be there.
So this is extended functionality.

"**Export plate as STL Individually**"

MNeMoNiCuZ, 2026-09-04:
This is still a requested feature, it's not been resolved by the 3.0.0 version.

frapaschal, 2026-09-14:
Please Prusa, make this a thing