prusa3d/PrusaSlicer #8637: Keyboard Shortcut for Export as STL...

WhiteDragonD, 2022-08-05:
### Description of the bug

I didn't see anything in the ? or options for a key mapping to Export as STL....

This is related to bug #8636.

### Project file & How to reproduce

Try to use the keyboard for Export as STL.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

Windows 10

### Printer model

I3 Mk2.5s

WhiteDragonD, 2022-08-05:
Windows 10 has a built in Shift + F10 option to execute a right click by default but it does not work in Slicer.

Valthol, 2024-02-06:
@WhiteDragonD OH MAN, look at my today's issue #12201 

Sad that noone is giving any interest for such important things for more demanding users. Have you managed to get any automations? I am doing it the dumb way with AutoHotKey with mouse clicks on fixed coordinates but the contex menu is a must have for me and it does not work :) Tried to overcome this, but "Export as stl" is only avaiable in the context menu :D ehh....

WhiteDragonD, 2024-05-09:
I have a Logitech keyboard that has some macro scripts that do help a bit... but you have to make sure you are setup properly before executing.  After opening up a set of files, lets say 20, I have to rotate and whatever modification I do then have to right click and export every single file..... 

So the script starts with my mouse over the first one, then start macro, it'll right click, do the E for export then Y to overwrite etc... if I had a 'command' to do the export then I could just use like a down arrow and then execute the export automatically.

> @WhiteDragonD OH MAN, look at my today's issue #12201
>
> Sad that noone is giving any interest for such important things for more demanding users. Have you managed to get any automations? I am doing [...]

closed 2026-09-29 by prusaslicer-bot[bot]