prusa3d/PrusaSlicer #8638: Prusa MINI+ network printer configuration disappears after Prusa Slicer restart

Tommy544, 2022-08-05:
### Description of the bug

I flashed the newest 4.4.0-BETA2 firmware on my Prusa MINI+ and I was able to add the printer as a network printer in Prusa Slicer (I tried 2.4.2, 2.5.0-ALPHA2 and 2.5.0-ALPHA3). The network communication seems to work fine and I was able to upload a .gcode file and start a print all from Prusa Slicer.

The problem is that every time I close Prusa Slicer (any version), the next time I start it the network printer configuration is gone. Am I missing some sort of Save/Store configuration button or is this not working as it should?

### Project file & How to reproduce

Not relevant

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10 64 bit

### Printer model

Prusa MINI+

FidelCapo, 2022-08-08:
Can you please describe the steps how you add network settings to the printer?
Thank you

Tommy544, 2022-08-16:
> 

Sorry for a late reply.

To add the network printer, I open Prusa Slicer, next I go to "Printer Settings" tab, then click on the cogwheel labeled "Add physical printer" and then I fill out the IP address of the printer, and the API key that I can find on the Prusa MINI screen. After I complete this wizard, I can see the physical printer added into the list of printers and when I slice a model, I can see and click the button that appears at the bottom right corner of Prusa Slicer, which sends the gcode to the printer and the printer is able to successfully start the print.

However, when I close Prusa Slicer, I lose this physical printer configuration and I have to go through the "Add physical printer" wizard again in order to be able to print over network again.

closed 2026-09-29 by prusaslicer-bot[bot]