prusa3d/PrusaSlicer #8640: Keyboard shortcuts are broken after clicking into sidebar.

WhiteDragonD, 2022-08-05:
### Description of the bug

If if press A (arrange) when the mouse is hovering in the Plate window... it will arrange.

If I click into the sidebar window, Pressing A (while hovering mouse in that window) A will NOT WORK.
If I click into the sidebar window, then move my mouse over the Plate (not clicking) then A will work.  
       This doesn't work for my needs because macros don't move the mouse.

If I click into the sidebar window then press Cntrl +1, the Plate window is not 'Active' so A does NOT WORK.
If I click into the sidebar window then press Shift + Tab to hide it.. it does NOT HIDE the sidebar.

### Project file & How to reproduce

see description

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

Windows 10

### Printer model

I3 Mk2.5s

WhiteDragonD, 2022-08-08:
Even if I am not using macros, this causes me to have to do an extra click (clicking back into the plate window) after clicking on the eye to make something printable instead of being able to press A for arrange.

stepikovo, 2022-08-22:
There should be no need to click, just hover over the plater with the cursor

closed 2026-09-29 by prusaslicer-bot[bot]