prusa3d/PrusaSlicer #8641: Prusa Slicer Freezes

LKastle, 2022-08-05:
### Description of the bug

When I open the program, before I even load a project file, I'm unable to switch between profiles (regardless of which dropdown I select (Print settings, Filament, and Printer), the dropdown will pop-up and the program becomes unresponsive.  I can go into any of the tabs (Print Settings, Filament Settings, or Printer Settings) and I can change any settings, so long as a dropdown menu is not involved.  IF I "tab" to the selection for a drop down, I can then switch between the options using the 'up' and 'down' arrow keys.

I've attempted running older versions.  I've tried uninstalling and reinstalling (v2.4.2).  Rebooting.  It was working fine last week, but has been giving me problems all this week.  

### Project file & How to reproduce

Open PrusaSlicer--> Click on any drop down in the app.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 11

### Printer model

Prusa MK3S (not relevant)

FidelCapo, 2022-08-08:
Unfortunately I am unable to reproduce this issue. Have you made any changes on your computer recently? Have you installed windows updates or graphics drivers? Can you please try version 2.5.0-alpha3?
Thank you

LKastle, 2022-08-08:
Hi FidelCapo.  The issue seems to have resolved itself over the weekend.  Cause and solution unknown.

hebbet, 2022-08-18:
i have the same issue with windows 11 with stable 2.4.2 and beta-1 of 2.5

Hammershoj, 2022-08-22:
I have the same issue with Windows 11 insider preview stable version 2.4.2, invoking any dropdowns result in an application hang

rtyr, 2022-08-22:
Does is happen with 2.5.0-rc1?

bzfvasco, 2022-09-26:
I have the same issue with Windows 11 but I'm using 2.5 (latest version). I also tried to run it with console and at startup the software shows "[error] Unable to parse version of AMD driver."
 Does anyone can help me?
(I've tried uninstalling and reinstalling)

lukasmatena, 2024-02-19:
This does not sound like PrusaSlicer related. Closing.

closed 2024-02-19 by lukasmatena