prusa3d/PrusaSlicer #8643: PrusaSlicer stores both state and settings in the same file breaking config sync and config management in git

VorpalBlade, 2022-08-05:
**Is your feature request related to a problem? Please describe.**

PrusaSlicer stores settings and state (recently opened files, version available online, window size, ...) in the same settings file `PrusaSlicer.ini` (on Linux in `~/.config/PrusaSlicer{,-alpha}/PrusaSlicer.ini`).

This causes issues if you try to sync settings files between computers in any automatic way (such as using a "dot file" manager like [chezmoi](https://www.chezmoi.io/), or via some other automatic sync program).

Here are some examples pulled from 2.5.0-alpha3+linux-x64-GTK3 config file:

```ini
# generated by PrusaSlicer 2.5.0-alpha3+linux-x64-GTK3 on 2022-08-05 at 20:44:08 UTC
; ...
last_output_path = /home/user/Desktop
last_output_path_removable = /run/media/user/3D_PRINTER
; ...
version_online = 2.4.2
version_online_seen = 2.4.2
version_system_info_sent = 2.5.0-alpha3+linux-x64-GTK3
; ...
window_mainframe = 910; 132; 926; 652; 0
; ...
[presets]
filament = Generic PETG
print = 0.20mm QUALITY @MK3
printer = Original Prusa i3 MK3S & MK3S+
; ...
[recent]
config_directory = /home/user/Documents/cad/myrecentproject
skein_directory = /home/user/Documents/cad/myrecentproject

[recent_projects]
1 = /home/user/Documents/cat/somefile.3md
2 = /home/user/Documents/cad/myrecentproject/somefile.3mf
```

These change far too often and/or are system specific for the file to be practical to sync.

**Describe the solution you'd like**

It would be preferable to store settings in one file and temporary state in another. 

**Describe how it would work**

PrusaSlicer should store actual user settings (things I change from the preferences window or configuration wizard for example) in one file. These are things that don't change often and users might want to sync between computers.

In another file should be ephemeral state: window size & position, most recently selected presets, recent files, results of online version checks, etc. These are things that change often and probably don't make sense to sync between computers anyway. Recent files for example are unlikely to be valid with the same paths on the other computer!

Another thing that should go away (from the synced file) is the comment at the top that is updated with the date the file is written every time. It needs to go away or only change when the file contents actually change.

On Linux the state file should optimally be put in a suitably named sub-directory to `$XDG_STATE_HOME` (default if unset, which it typically is: `$HOME/.local/state`). See the full [specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html). I have absolutely no idea where the file should go on Windows or Mac OS X, I don't think they have separate config and state directories?

**Describe alternatives you've considered**
I have in fact written a script to filter the settings file before syncing. This works but is an annoying workaround and specific to my setup. Anyone using a different syncing solution than me cannot take advantage of it.

**Additional context**
Add any other context or paint drawing about the feature request here.

alexiri, 2023-07-29:
This bit me recently while trying to work around the lack of sync functionality (ie. #4898). Please change this behavior!

VorpalBlade, 2023-07-29:
> This bit me recently while trying to work around the lack of sync functionality (ie. #4898). Please change this behavior!

I have a workaround: I manage my configuration files in git using [chezmoi](https://www.chezmoi.io/) for templating etc. I wrote an [add-on](https://github.com/VorpalBlade/chezmoi_modify_manager) to chezmoi to deal with these types of settings files (it isn't just PrusaSlicer doing things incorrectly).

Probably only a solution for people comfortable with working in the terminal and with version control, so it might not suit you. But it is an option.

alexiri, 2023-07-29:
I'm interested in seeing what you've done, but I'm not sure if I'll be able to use it. Unfortunately, I'd like to sync between a Linux and a Windows machine.

VorpalBlade, 2023-07-29:
> I'm interested in seeing what you've done, but I'm not sure if I'll be able to use it. Unfortunately, I'd like to sync between a Linux and a Windows machine.

I believe chezmoi supports Windows, and my addons builds in CI for Windows. I don't have a computer i can test it on though. One potential issue is that the paths relative to the home directory will be different. Chezmoi can handle symlinks (at least on Linux) so it might be possible to work around it.