prusa3d/PrusaSlicer #8644: program exits when I hit "Slice now"

epleace, 2022-08-06:
### Description of the bug

what happened: program exits when I hit "Slice now" with all default settings.
I expected: a slicing to generate gcode.

I've tried it a few times, with and without the settings I wanted.
(note, this web page will not accept my compressed binary .stl file, seems ok with compressed text .stl)

### Project file & How to reproduce

Compressed stl file:
[pulley wheel0.zip](https://github.com/prusa3d/PrusaSlicer/files/9274294/pulley.wheel0.zip)

===========================================================================
[image]
[pulley wheel.7z.zip](https://github.com/prusa3d/PrusaSlicer/files/9274284/pulley.wheel.7z.zip)

===========================================================================
PrusaSlicer
Version:   2.5.0-alpha3+win64
Build:     PrusaSlicer-2.5.0-alpha3+win64-202207151230

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 16,989MB
OpenGL installation
GL version:   4.6.14800 Compatibility Profile Context 22.6.1 30.0.21023.1015
Vendor:       ATI Technologies Inc.
Renderer:     Radeon RX 580 Series
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
[... 261 lines cut ...]
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 - alpha3

### Operating system

win 10

### Printer model

prusa i3 mk3s+

FidelCapo, 2022-08-08:
Can you please attach the project (File -> Save project as..)? Thank you

epleace, 2022-08-09:
Hello,
Thanks for the response.
Attached is the file.
Best wishes
Eric

benkrejci, 2022-09-07:
This also happens to me, but only with specific presets. I can reproduce on alpha3 and rc1

Here's the info for mine if it helps:

[gridfinity base plate 4x4.zip](https://github.com/prusa3d/PrusaSlicer/files/9501977/gridfinity.base.plate.4x4.zip)

Crash occurs when using the following presets (unmodified from latest default profiles downloaded via configuration wizard today 2022-09-06):
- Print setting: "0.35mm SPEED"
- Printer: "Original Prusa MINI & MINI+ 0.6 nozzle"
- Filament: any (e.g. Prusament PLA)

PrusaSlicer
Version:   2.5.0-rc1+win64
Build:     PrusaSlicer-2.5.0-rc1+win64-202208191450

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 34,301MB
OpenGL installation
GL version:   4.6.0 NVIDIA 516.59
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3070 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
[... 334 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

kubispe1, 2023-04-06:
@benkrejci and @epleace I hope this issue has been already fixed in 2.6.0-alfas.
There are .stls and no .3mf, so it is not possible to reproduce it. I am closing this issue. If you guys find it is not resolved, let me know.

closed 2023-04-06 by kubispe1