prusa3d/PrusaSlicer #8645: PrusaSlicer crashes after latest Ubuntu updates

linuxdudeg, 2022-08-06:
### Description of the bug

As soon as I try to start PrusaSlicer I get the following error:
[2022-08-06 14:25:12.287237] [0x00007f72187618c0] [trace]   Initializing StaticPrintConfigs

(prusa-slicer:26866): Gtk-WARNING **: 14:25:12.678: Theme directory places/symbolic of theme tuxedo-breeze has no size field

(prusa-slicer:26866): Gdk-ERROR **: 14:25:12.733: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 354 error_code 2 request_code 152 (GLX) minor_code 3)

### Project file & How to reproduce

Try to run any of the Linux AppImage files. They were working fine until the latest Ubuntu update

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2, 2.5.0-alpha2, 2.5.0-alpha3

### Operating system

Ubuntu 22.04

### Printer model

Ender 3 Pro, Prusa i3mks+

Sonicboom247, 2022-08-06:
odd, i am running Ubuntu 22.04 and never had Prusa 2.5.0-alpha2 or alpha3 crash. I am just running from the extracted zip file.

linuxdudeg, 2022-08-07:
Sorry this was my mistake, I didn't restart the laptop after the last updates. Everything working fine this morning

closed 2022-08-07 by linuxdudeg