prusa3d/PrusaSlicer #8648: Lightning infill print on air

ilanodem, 2022-08-07:
### Description of the bug

I've printed several models using the new lightning infill which printed just fine. However, on a certain duck model the infill generation seems to be wrong and some of it just prints on air. 

### Project file & How to reproduce

So far the model in the attached project is the only one I can reproduce the error on. It seems to be happening regardless of settings such as infill percentage, walls, top/bottom layers, flow, widths, speeds, etc.

Stl file:
[Rubber_Duck.zip](https://github.com/prusa3d/PrusaSlicer/files/9276646/Rubber_Duck.zip)

[image]
.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Alfawise U20 (Heavily modified)

hejllukas, 2022-08-08:
I wasn't able to reproduce that on your model (STL). Could you please upload 3MF project file (File->Save Project as...)?

ilanodem, 2022-08-09:
Sure, here it is:
[Rubber_Duck_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9287385/Rubber_Duck_3mf.zip)

foreachthing, 2022-08-09:
Got it: "Only infill when needed" has to be off/false.
See here: Print Settings -> Infill -> Reducing print time

ilanodem, 2022-08-09:
You're right, thanks! It's still strange that other models generated correct infill even when true.
In any case maybe a warning should be added that this setting better be disabled when using lightning infill.

jb221, 2022-08-16:
> In any case maybe a warning should be added that this setting better be disabled when using lightning infill.

Or have lightning infill ignore the setting and treat it as off, since the point of lightning infill is redundant to that setting.

hejllukas, 2022-08-18:
@foreachthing Thank you for investigating this issue.

As @jb221 suggested, we let Lightning infill ignore the setting "Only infill where needed". There isn't much sense for combining those settings because Lightning could be viewed as equivalent to "Only infill where needed".

It was fixed with https://github.com/prusa3d/PrusaSlicer/commit/6f5813a849e40b1bcae97270a317222dff92f5ec in master_250, and it will be part of the next release.

closed 2022-08-18 by hejllukas