prusa3d/PrusaSlicer #8649: Displacement mapping a.k.a. bump mapping at the slicing stage could be leveraged to provide paint-on fuzzy and fuzzy features and more.

poikilos, 2022-08-07:
**Is your feature request related to a problem? Please describe.**
Lol "I'm always frustrated when" the model I download is dozens of megabytes because the author had to apply a displacement map to the mesh with automatic subdivision, yet when I decimate it to 1/20 of its original file size the model isn't visibly different.

**Describe the solution you'd like**
I would like to add a displacement map during the slicing stage.

**Describe how it would work**
For each island on a layer, the distance from the center of that island would be used to determine an arc length to map back to the texture using a pixels_per_mm value (that the user could set). For the vertical mapping, the z coordinate of the bottom of the layer + 1/2 of the layer height would be used.
- Using the arc length of a virtual circle it for consistency: It will ensure that the texture mapping doesn't have zigzag patterns when the path zigzags.

**Describe alternatives you've considered**
Adding a displacement map to the model may be more of a standard process and have shared code and respect original UV maps. However, this would only work for models designed with mapping in mind. I know how to add a UV map myself and redo it to be tidy and even, but most people do not. Therefore, doing the process using math is more useful to more people. Doing it at the time of slicing will also save RAM (even in the slicing preview, only more segments not more triangles would be stored, so it would use geometrically less RAM).

**Additional context**
- At my displacement_map branch I will try to implement the feature myself before asking for help, by forking from the 2021 vb_fuzzy_skin branch: https://github.com/prusa3d/PrusaSlicer/tree/vb_fuzzy_skin as follows.

Tasks:
- [x] fork vb_fuzzy_skin branch as displacement_map branch
```
# forked to <https://github.com/poikilos/PrusaSlicer>
# then as per <https://blog.scottlowe.org/2015/01/27/using-fork-branch-git-workflow/>
git remote add upstream https://github.com/prusa3d/PrusaSlicer.git
# then as per <https://github.com/orgs/community/discussions/23445>:
git fetch upstream vb_fuzzy_skin && git checkout vb_fuzzy_skin

git branch displacement_map
git checkout displacement_map
git push --all
```
- [ ] Add settings and relevant UI widgets for choosing: displacement_map_path, pixels_per_mm, and displacement_mm (depth)
- [ ] Load the map and read a pixel's brightness from it, showing test output on the console to that effect.
- [ ] Map the texture to the outer wall path of each island separately.

Poikilos, 2022-12-24:
Progress! 
[image]

Poikilos, 2022-12-24:
Tasks:
- [ ] Fix alignment
[image]

Poikilos, 2022-12-28:
It calculates normals! Cube mapping is complete!
[image]

Poikilos, 2022-12-28:
With this 1024x1024 texture and large model, it seems to hang at 90% though the Export option is available. After a few minutes it says Exporting.
- https://ambientcg.com/view?id=Bricks076A
- https://www.thingiverse.com/thing:4077316

adenta, 2024-02-06:
So what ever came of this?

Poikilos, 2024-03-21:
> So what ever came of this?

You tell me...It works fine but no one noticed. Recently there was more discussion of this feature not existing here: [3D Print Any Texture You Want!](https://www.youtube.com/watch?v=MezOnZN1x18) on YouTube

adenta, 2024-03-21:
@Poikilos what do you mean noone noticed? Can we bundle this with Orca/Bambu perhaps?

Poikilos, 2024-03-22:
Well, I mean I thought more people might find, react, and discuss this issue, and maybe maintainers would put it up for discussion, since people talk about the feature a lot just not on GitHub :)

adenta, 2024-03-22:
I want it!

Chocolateandmilkwin, 2024-09-09:
Sad that there isn't a star feature for issues, this would be such a huge addition for micro detailing.

Poikilos, 2024-09-09:
> Sad that there isn't a star feature for issues, this would be such a huge addition for micro detailing.

Well, you could react to the original post by clicking emoji then thumbs up, or in this case, click the thumbs up that is already there. Typically people add "thumbs up" on issues on GitHub, so maybe if enough people do, it will get traction.

Keavon, 2024-12-26:
@Poikilos I'm excited for the possibility of this feature landing but I'm confused... I don't see a PR linked anywhere in this issue. Did you open one? Shouldn't that be where the discussion and hoping-to-get-noticed-by-maintainers occurs?

adenta, 2024-12-26:
There is a formlabs tool that does a half decent job

Poikilos, 2024-12-26:
> There is a formlabs tool that does a half decent job

That one operates on the model, so it doesn't solve the stated problem. The page "Introducing the Formlabs Texture Engine" says "ensure that the mesh density is high" which is the problem I stated in my issue and I solved. Many tools can do that (including applying subdivision and displacement map modifiers in Blender). On the other hand, you can run mine on a cube (8 vertices) and only the G-code will contain the displacement.

Keavon, 2024-12-26:
@Poikilos but do you have a PR open for your implementation?

Poikilos, 2024-12-27:
If I added a feature in a pull request I didn't know how well that fits with the expectations here, but if that's the way then fine. It needs to be rebased but I should be able to do that. Or I could just add it anyway and conflicts can be resolved during merge. That may not be enough since the change will probably have to be added to the Arachne code if that is separate.

Keavon, 2024-12-27:
I'd highly encourage you to rebase it and open a PR. Since you've already put in the effort to build most of the feature, that is certainly the next step in getting it considered for review in an open source environment. It's no surprise that this issue hasn't been seen by of the maintainers— there are 4.5k open issues compared to just 168 open PRs. I doubt the maintainers are following this one particular issue, but they'd be more likely to look at the incoming PRs which is the avenue for contributing actual code changes (issues aren't). As for Arachne, perhaps it's better to note that in the PR and see what the maintainers propose. I look forward to seeing this move forward via the proper pathways (a PR)!

Keavon, 2025-01-07:
Hi @Poikilos, just checking in about your plans to open a PR for this. I hope you do it, because it's an important first step towards making this go further. Thank you!

Poikilos, 2025-01-07:
Thanks for your interest. I did add a PR just to get some eyes on it:  #13854 but I need to rebase it manually unless someone can help with that. It is a significant change because that part of the code is now abstracted (the slicing function with the bulk of my code now has `_classic` appended to its name, apparently), possibly related to Arachne being an optional slicer now (PR is new but commits in my branch are from ~1yr ago). I was hoping someone would notice who has an understanding of the code changes and offer help or guidance. That doesn't seem to be happening.

Again I'm mystified why everyone talks about this feature (on YouTube channels and forums, discussing multiple slicers) except not on actual repos. I coded it with no knowledge of the codebase last time, so I guess I'll have to do it again. I have a slight head start since I know which arguments need to be added to carry the vertex data to the final step, but some arguments of even those functions differ now in master vs when I started. Also, the fuzzy skin branch I based on was apparently rebased then merged, because it is in the code but differs greatly. I will have to get back to this after a game jam I am in until Jan 17th.

undingen, 2025-01-09:
Hi thanks for this awesome feature: I modified your code make it work in current OrcaSlicer and put it up as a draft. It contains a punch of hacks but would like to get more feedback before putting more effort into it...
Could you please take a look: https://github.com/SoftFever/OrcaSlicer/pull/7985

Poikilos, 2026-09-14:
> This issues has been labeled **legacy**. It means that it has been created before the release of _PrusaSlicer 3.0.0 alpha_ version. In an attempt to clean up the GitHub issues tracker any **legacy** [...]

This feature is still not in most slicers, and ones that do implement it now (or have forks that do so) other than undingen's OrcaSlicer implementation do so not at the slicing stage but with geometry which is has the storage and throughput problems (The feature gained there is UV control and previews, but modeling programs are already better at that...if you want a solution using all that storage and throughput).