prusa3d/PrusaSlicer #8650: "Snug" supports generate in midair

dimo414, 2022-08-08:
### Description of the bug

[image]

As the screenshot shows, using "snug" supports generates these floating artifacts instead of usable support structures.

### Project file & How to reproduce

[SnugBug.zip](https://github.com/prusa3d/PrusaSlicer/files/9280542/SnugBug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa MINI+

FlipAirboost, 2022-08-15:
Un peu la même chose chez moi. J'ai des supports qui s'impriment à l'extérieur sur la pièce. 
Ils ne sont pas visibles sur la modélisation.
Ce bug apparait quand je change espace motif et support.
Ils sont faciles à détacher, preuve que la distance de séparation  XY et Z est respectée.
[image]

kubispe1, 2022-08-17:
same as #7816

closed 2026-09-29 by prusaslicer-bot[bot]